platformio/platform-atmelsam #4: Serial USB not working

stonebyte, 2017-01-29:
Hi,

Problem observed: 
- Serial USB output is not working with platformio. With Arduino IDE it is working as expected.

Test setup:
- Board: Adafruit Feather M0 (atmelsam)
- Platformio: 
IDE 1.7.1 | Core 3.2.1 under MacOS and Linux
```
Platform atmelsam
--------
Updating atmelsam @ 1.6.0: 	[Up-to-date]
Updating framework-arduinosam @ 1.10611.0: 	[Up-to-date]
Updating toolchain-gccarmnoneeabi @ 1.40804.0: 	[Up-to-date]
Updating tool-bossac @ 1.10601.0: 	[Up-to-date]
```

Simple test sketch:
```
#include <Arduino.h>
int n=0;

void setup()
{
   Serial.begin(9600);
}

void loop()
{
   Serial.println(n++);
}
```

Compiling and uploading this sketch using the Arduino GUI yields the expected result (fast running list of numbers) in the serial console.

Compiling and uploading using platformio in general works as well:

```
pio run -t upload
[Sat Jan 28 20:00:07 2017] Processing adafruit_feather_m0_usb (platform: atmelsam, board: adafruit_feather_m0_usb, framework: arduino)
--------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 6 compatible libraries
Looking for dependencies...
Project does not have dependencies
Linking .pioenvs/adafruit_feather_m0_usb/firmware.elf
Checking program size
text	   data	    bss	    dec	    hex	filename
9712	    256	   1784	  11752	   2de8	.pioenvs/adafruit_feather_m0_usb/firmware.elf
Looking for upload port...
Auto-detected: /dev/cu.usbmodem1D1151
Forcing reset using 1200bps open/close on port /dev/cu.usbmodem1D1151
Waiting for the new upload port...
Uploading .pioenvs/adafruit_feather_m0_usb/firmware.bin
Atmel SMART device 0x10010005 found
Erase flash
done in 0.627 seconds

Write 9968 bytes to flash (156 pages)
[==============================] 100% (156/156 pages)
done in 0.067 seconds

Verify 9968 bytes of flash with checksum.
Verify successful
done in 0.007 seconds
CPU reset.
=========================================== [SUCCESS] Took 5.42 seconds ===========================================
```

**_But_: Theres no output in the serial console.**

Everything else seems to work fine if I upload code which does other things like blinking a LED.

jacobstim, 2017-01-30:
Try using 'SerialUSB' instead of Serial.  Struggled with the same issue after upgrading PlatformIO for all my Adafruit Feather boards...

As a tip, use "#define OUTSERIAL SerialUSB" at the beginning of all your sketches and use OUTSERIAL.print("My output") in the sketch; that allows easy redirection towards other outputs.

stonebyte, 2017-01-30:
Thanks @jacobstim!

That was a very useful hint and solved the problem. 

Wouldn't have tried it without your hint as the Adafruit [howto](https://learn.adafruit.com/adafruit-feather-m0-basic-proto/adapting-sketches-to-m0#serial-vs-serialusb) for the Feather M0 explicitly states not to use 'SerialUSB' but 'Serial'. 

Replacing the Feather M0 board variant with the Arduino Zero variant under the hood is a quite unfortuante decision which leads to confusion.

ivankravets, 2017-01-30:
You are tight, we use the latest Arduino Core for all SAMD boards. We have 2 options:

1. Leave as it is, use the latest Core/HAL for Adafruit Feather M0 as we do that for Zero, etc.
2. Switch to old Adafruit's SAMD core and link Adafruit Feather M0 with it.

I don't know what is better.

jacobstim, 2017-01-31:
Having looked briefly at the Adafruit M0 variant/cores in the past months, I got the impression that there is a 99,9% overlap with the standard Arduino Zero variant/core, with only minor modifications such as SerialUSB being mapped to Serial, etc. So from that very limited experience, I would be inclined to suggest to standardize to a single SAMD Arduino Core. That seems to fit better in the "philosophy of PlatformIO" (or at least the perception I have of it, being an outsider to the project), to standardize & introduce code portability, regardless of the underlying MCU board. 

I'll even go one step further... I am mixing in my prototyping the Sparkfun SAMD21 breakouts with Arduino MKR1000's and Adafruit Feather M0's, and I found myself constantly redefining the "serial" output port when switching between boards, because of the (IMHO unnecessary) discrepancies between the different variants/cores. So for me personally it's a step forward to standardize on a single SAMD core. 

The downside is, as @stonebyte mentions, the user experience. At least some documentation should be written that explains the difference?

stonebyte, 2017-01-31:
I support @jacobstim view that adding some sort of documentation would help if such things can't be prevented otherwise. 
Perhaps adding a link to a page with caveats for boards in the http://platformio.org/boards page would do the job.

deladriere, 2017-01-31:
Ok to standardize for the serial port but I just lost the analog A6 A7 from the feather M0 !
I badly need them. How  can I get them back? Do you have a similar work around so I can access A6 within the Zero ?

jacobstim, 2017-02-01:
Regarding the documentation specifically for the Adafruit Feather M0 boards, they mention that you can use SerialUSB as well, see here: 
https://learn.adafruit.com/adafruit-feather-m0-basic-proto/adapting-sketches-to-m0#serial-vs-serialusb

Regarding the A6 and A7 pins... well spotted! The Arduino Zero core defines up to pin A5, see here:
https://github.com/arduino/ArduinoCore-samd/blob/master/variants/arduino_zero/variant.h

Indeed, the Adafruit variation of the Arduino Zero core adds a few pins, see e.g. "variants/arduino_zero/variant.h" in the official Adafruit Feather M0 boards for the Arduino IDE: https://adafruit.github.io/arduino-board-index/boards/adafruit-samd-1.0.9.tar.bz2

I think you can work around this temporarily by adding the following to your code:

#define PIN_A6               (8ul)
#define PIN_A7               (9ul)
static const uint8_t A6  = PIN_A6 ;
static const uint8_t A7  = PIN_A7 ;

deladriere, 2017-02-01:
@jacobstim Thanks a lot for the #define suggestion !
It does compile but doesn't work: I have a pot connected to A6 but still can't read the analog value

deladriere, 2017-02-01:
I also added
 #define NUM_ANALOG_INPUTS    (8u)  
but still it doesn't work

stonebyte, 2017-02-01:
@jacobstin Actually the Adafruit documentation made me think that I had to use Serial, not SerialUSB. I wasn't aware that platformio is using the generic SAMD core and not the feather_M0 specific variant. 

From my point of view staying with the latest SAMD core and giving some guidance to the users in documentation should do the job.

jacobstim, 2017-02-01:
@deladriere Oh, I overlooked that in variant.cpp the digital pins 8 and 9 also have the 'analog' attribute (PIN_ATTR_ANALOG) set in the Adafruit M0 board variant, which I think enables the reading of analog values. That makes it certainly messier to work around temporarily :). 

Perhaps it is better to use the old version (1.5.0) for now? I read in a similar issue report that this should work: https://github.com/platformio/platform-atmelsam/issues/5

deladriere, 2017-02-02:
@jacobstim how can I revert to 1.5.0 ? Because now I have to use the Arduino IDE to work on my code 
Maybe it would be more fair to consider removing the feather_M0 from PIO ?

ivankravets, 2017-02-02:
Guys, I need your help with this board. Try this:

1. Download https://github.com/adafruit/arduino-board-index/blob/gh-pages/boards/adafruit-samd-1.0.13.tar.bz2?raw=true
2. Unpack
3. Rename `~/.platformio/packages/framework-arduinosam/variants/arduino_zero` to `~/.platformio/packages/framework-arduinosam/variants/arduino_zero_backup`
4. Copy unpacked folder `adafruit-samd-1.0.13/variants/arduino_zero` to `~/.platformio/packages/framework-arduinosam/variants/`.
5. Run `pio run -t clean`
6. Build project and test firmware.

If it works, we will create separate variant for this board and update development platform.

commit 9d3c7afb (2017-02-02): Use SAMD legacy core for Adafruit Feather M0 and SODAQ boards // Resolve #4 Resolve #5 Resolve #6
closed 2017-02-02 by ivankravets