platformio/platform-atmelsam #6: fatal error: variant.h: No such file or directory

jwillemsen, 2017-02-01:
I am trying to enable travis-ci for an arduino related effort. Created a travis-ci yml file at https://github.com/jwillemsen/RN2483-Arduino-Library/blob/master/.travis.yml and enabled it on travis-ci. This now results in

```
Setting up build cache

$ export CASHER_DIR=$HOME/.casher

0.03s$ Installing caching utilities

0.00s

1.71sattempting to download cache archive

fetching master/cache-linux-precise-8b38933858bcbc861bcd2286a72dd89c26b411e0527c0350b7ca7ca4748667b7--python-2.7.tgz

found cache

0.00s

5.79sadding /home/travis/.platformio to cache

creating directory /home/travis/.platformio

$ python --version

Python 2.7.9

$ pip --version

pip 6.0.7 from /home/travis/virtualenv/python2.7.9/lib/python2.7/site-packages (python 2.7)
install

3.06s$ pip install -U platformio

Collecting platformio

  Downloading platformio-3.2.1-py27-none-any.whl (120kB)

    100% |################################| 122kB 3.5MB/s 

Collecting bottle<0.13 (from platformio)

  Downloading bottle-0.12.13.tar.gz (70kB)

    100% |################################| 73kB 5.5MB/s 

Collecting requests<3,>=2.4.0 (from platformio)

  Downloading requests-2.13.0-py2.py3-none-any.whl (584kB)

    100% |################################| 585kB 886kB/s 

Collecting pyserial<4,>=3 (from platformio)

  Downloading pyserial-3.2.1-py2.py3-none-any.whl (189kB)

    15% |####          100% |################################| 192kB 2.7MB/s 

Collecting semantic-version>=2.5.0 (from platformio)

  Downloading semantic_version-2.6.0.tar.gz

Collecting colorama (from platformio)

  Downloading colorama-0.3.7-py2.py3-none-any.whl

Collecting click<6,>=5 (from platformio)

  Downloading click-5.1-py2.py3-none-any.whl (65kB)

    100% |################################| 65kB 6.3MB/s 

Collecting lockfile<0.13,>=0.9.1 (from platformio)

  Downloading lockfile-0.12.2-py2.py3-none-any.whl

Installing collected packages: lockfile, click, colorama, semantic-version, pyserial, requests, bottle, platformio

  Running setup.py install for semantic-version

  Running setup.py install for bottle

    changing mode of build/scripts-2.7/bottle.py from 644 to 755

    changing mode of /home/travis/virtualenv/python2.7.9/bin/bottle.py to 755

Successfully installed bottle-0.12.13 click-5.1 colorama-0.3.7 lockfile-0.12.2 platformio-3.2.1 pyserial-3.2.1 requests-2.13.0 semantic-version-2.6.0

2.82s$ platformio ci $PLATFORMIO_CI_EXTRA_ARGS

The next files/directories have been created in /tmp/tmp2GHx1h

platformio.ini - Project Configuration File

src - Put your source files here

lib - Put here project specific (private) libraries

Project has been successfully initialized!

Useful commands:

`platformio run` - process/build project from the current directory

`platformio run --target upload` or `platformio run -t upload` - upload firmware to embedded board

`platformio run --target clean` - clean project (remove compiled files)

`platformio run --help` - additional information

[Wed Feb  1 12:15:11 2017] Processing sodaq_one (platform: atmelsam, board: sodaq_one, framework: arduino)

--------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option

Converting SodaqOne-TTN-Mapper-ascii.ino

Collected 6 compatible libraries

Looking for dependencies...

Library Dependency Graph

|-- <Wire> v1.0

Compiling .pioenvs/sodaq_one/src/SodaqOne-TTN-Mapper-ascii.ino.o

Archiving .pioenvs/sodaq_one/libFrameworkArduinoVariant.a

Compiling .pioenvs/sodaq_one/src/Sodaq_UBlox_GPS.o

In file included from /home/travis/.platformio/packages/framework-arduinosam/cores/arduino_samd/Arduino.h:81:0,

from /tmp/tmpf1IhZw:1:

/home/travis/.platformio/packages/framework-arduinosam/cores/arduino_samd/delay.h:27:21: fatal error: variant.h: No such file or directory

#include "variant.h"

^

compilation terminated.

Indexing .pioenvs/sodaq_one/libFrameworkArduinoVariant.a

In file included from /home/travis/.platformio/packages/framework-arduinosam/cores/arduino_samd/Arduino.h:81:0,

from src/Sodaq_UBlox_GPS.cpp:20:

/home/travis/.platformio/packages/framework-arduinosam/cores/arduino_samd/delay.h:27:21: fatal error: variant.h: No such file or directory

#include "variant.h"

^

compilation terminated.

Compiling .pioenvs/sodaq_one/FrameworkArduino/IPAddress.o

Compiling .pioenvs/sodaq_one/FrameworkArduino/Print.o

Compiling .pioenvs/sodaq_one/FrameworkArduino/Reset.o

Compiling .pioenvs/sodaq_one/FrameworkArduino/RingBuffer.o

In file included from /home/travis/.platformio/packages/framework-arduinosam/cores/arduino_samd/Arduino.h:81:0,

from /home/travis/.platformio/packages/framework-arduinosam/cores/arduino_samd/IPAddress.cpp:20:

/home/travis/.platformio/packages/framework-arduinosam/cores/arduino_samd/delay.h:27:21: fatal error: variant.h: No such file or directory

#include "variant.h"

^

compilation terminated.

Compiling .pioenvs/sodaq_one/FrameworkArduino/SERCOM.o

In file included from /home/travis/.platformio/packages/framework-arduinosam/cores/arduino_samd/Arduino.h:81:0,

from /home/travis/.platformio/packages/framework-arduinosam/cores/arduino_samd/Print.cpp:23:

/home/travis/.platformio/packages/framework-arduinosam/cores/arduino_samd/delay.h:27:21: fatal error: variant.h: No such file or directory

#include "variant.h"

^

compilation terminated.

Compiling .pioenvs/sodaq_one/FrameworkArduino/Stream.o

In file included from /home/travis/.platformio/packages/framework-arduinosam/cores/arduino_samd/Arduino.h:81:0,

from /home/travis/.platformio/packages/framework-arduinosam/cores/arduino_samd/Reset.cpp:19:

/home/travis/.platformio/packages/framework-arduinosam/cores/arduino_samd/delay.h:27:21: fatal error: variant.h: No such file or directory

#include "variant.h"

^

compilation terminated.

Compiling .pioenvs/sodaq_one/FrameworkArduino/Tone.o

/home/travis/.platformio/packages/framework-arduinosam/cores/arduino_samd/SERCOM.cpp:20:21: fatal error: variant.h: No such file or directory

#include "variant.h"

^

compilation terminated.

Compiling .pioenvs/sodaq_one/FrameworkArduino/USB/CDC.o

Compiling .pioenvs/sodaq_one/FrameworkArduino/USB/PluggableUSB.o

In file included from /home/travis/.platformio/packages/framework-arduinosam/cores/arduino_samd/Arduino.h:81:0,

from /home/travis/.platformio/packages/framework-arduinosam/cores/arduino_samd/Stream.cpp:25:

/home/travis/.platformio/packages/framework-arduinosam/cores/arduino_samd/delay.h:27:21: fatal error: variant.h: No such file or directory

#include "variant.h"

^

compilation terminated.

Compiling .pioenvs/sodaq_one/FrameworkArduino/USB/USBCore.o

In file included from /home/travis/.platformio/packages/framework-arduinosam/cores/arduino_samd/Arduino.h:81:0,

from /home/travis/.platformio/packages/framework-arduinosam/cores/arduino_samd/Tone.h:23,

from /home/travis/.platformio/packages/framework-arduinosam/cores/arduino_samd/Tone.cpp:19:

/home/travis/.platformio/packages/framework-arduinosam/cores/arduino_samd/delay.h:27:21: fatal error: variant.h: No such file or directory

#include "variant.h"

^

compilation terminated.

Compiling .pioenvs/sodaq_one/FrameworkArduino/USB/samd21_host.o

In file included from /home/travis/.platformio/packages/framework-arduinosam/cores/arduino_samd/Arduino.h:81:0,

from /home/travis/.platformio/packages/framework-arduinosam/cores/arduino_samd/USB/CDC.cpp:19:

/home/travis/.platformio/packages/framework-arduinosam/cores/arduino_samd/delay.h:27:21: fatal error: variant.h: No such file or directory

#include "variant.h"

^

compilation terminated.

Compiling .pioenvs/sodaq_one/FrameworkArduino/Uart.o

Compiling .pioenvs/sodaq_one/FrameworkArduino/WInterrupts.o
```

jwillemsen, 2017-02-01:
Reported it here as suggested by https://community.platformio.org/t/missing-variant-h-and-sodaq-board/1325

michaelarfreed, 2017-03-30:
I'm having the same issue (using the Adafruit Feather M0 board). @ivankravets can you clarify how to solve this?

DavidAntliff, 2020-01-07:
I also see this with a Sparkfun M0 board (Razor M0, based on the SparkFun SAMD21 Mini Breakout board definition .json):

```
Compiling .pio/build/sparkfun_9dof_razor_imu_m0/src/Compass.cpp.o
In file included from /home/david/.platformio/packages/framework-arduinosam/cores/samd/Arduino.h:81:0,
                 from src/Compass.cpp:3:
/home/david/.platformio/packages/framework-arduinosam/cores/samd/delay.h:23:10: fatal error: variant.h: No such file or directory
```

The original project (before I renamed the `.ino` files and added `#include <Arduino.h>` to them) compiles fine in the Arduino IDE, so I'd really like to know how to get this working with PlatformIO.

commit 9d3c7afb (2017-02-02): Use SAMD legacy core for Adafruit Feather M0 and SODAQ boards // Resolve #4 Resolve #5 Resolve #6