platformio/platform-atmelsam #9: i2c hangs  using mbed on Samd21 xplain pro board

jorgeorta, 2017-02-15:
I think there is something wrong... i can make a simple serial port print on a while loop every second.. but as soon as i add I2C i2c(PA08,PA09) it stops working and i dont get anything on serial port.

jorgeorta, 2017-03-09:
`#include "mbed.h"
#include "string.h"

//--------Board pin Declarations
DigitalOut myled(PB30);
Serial pc(PA22,PA23);//tx, rx
DigitalIn interrupt1(PB04);
DigitalIn interrupt2(PB05);
I2C i2c(PA08,PA09);// I2C pins on Xplained Pro Board

char deviceAddress = 0x1D;

int main()
{
  char byte_read[1];
  char who_am_i[1];
  who_am_i[0]=0xD;
  byte_read[0]=0xFF;
while(1)
  {

        i2c.start();                 // Start
        i2c.write(deviceAddress,who_am_i,1,true);
        i2c.stop();
        // done

    wait_ms(500);
    myled=1;
    wait_ms(500);
    myled=0;

  }

}`

so if i use this code, in theory i should be writting two times on the SDA line isnt it? well on my scope i am only seeing one clock burst. so it is only sending the device address and it missed sending the register i want to read.

[http://i66.tinypic.com/2zewmk9.jpg](url)

ivankravets, 2017-03-09:
Does it work with mbed online IDE?

ivankravets, 2017-04-01:
We have updated mbed framework. Do you still experience this problem?

ivankravets, 2018-03-13:
Microchip left ARM mbed and ARM dropped support for this board. Please use Arduino framework instead.

closed 2018-03-13 by ivankravets