platformio/platform-atmelsam #10: Upload Error on sodaq_autonomo : SAM-BA operation failed

loicguillaume, 2017-02-23:
Hi, I'm trying to upload a program to my sodaq_autonomo but unfortunatly it fails.

The build is fine, but when i'm trying to upload the program:

```
Looking for upload port...
Auto-detected: COM7
Forcing reset using 1200bps open/close on port COM7
Waiting for the new upload port...
Uploading .pioenvs\sodaq_autonomo\firmware.bin
 
SAM-BA operation failed
Atmel SMART device 0x10010000 found
*** [upload] Error 1
 [ERROR] Took 27.73 seconds
```

I saw a similar error have been reported here: https://community.platformio.org/t/unable-to-load-to-mkrzero/1502

For the record, here is the output with the Arduino IDE (I need to press the reset button to start the upload):

```
Atmel SMART device 0x10010000 found
Device       : ATSAMD21J18A
Chip ID      : 10010000
Version      : v2.0 [Arduino:XYZ] May  9 2016 16:06:46
Address      : 8192
Pages        : 3968
Page Size    : 64 bytes
Total Size   : 248KB
Planes       : 1
Lock Regions : 16
Locked       : none
Security     : false
Boot Flash   : true
BOD          : true
BOR          : true
Arduino      : FAST_CHIP_ERASE
Arduino      : FAST_MULTI_PAGE_WRITE
Arduino      : CAN_CHECKSUM_MEMORY_BUFFER
Erase flash
done in 0.844 seconds

Write 14376 bytes to flash (225 pages)

[========                      ] 28% (64/225 pages)
[=================             ] 56% (128/225 pages)
[=========================     ] 85% (192/225 pages)
[==============================] 100% (225/225 pages)
done in 0.089 seconds

Verify 14376 bytes of flash with checksum.
Verify successful
done in 0.018 seconds
CPU reset.
```

Regards.

joscha, 2017-03-05:
Same for the SODAQ ONE @loicguillaume, a Sketch looking as follows:

```c
void setup() {
   pinMode(ENABLE_PIN_IO, OUTPUT);
   digitalWrite(ENABLE_PIN_IO, HIGH);

  SerialUSB.begin(115200);
  SerialUSB.println("Hello");
}

void loop() {
  SerialUSB.println("World");
}
```
compiles and uploads fine with the Arduino tool:
```
Sketch uses 8456 bytes (3%) of program storage space. Maximum is 262144 bytes.
Atmel SMART device 0x10010005 found
Device       : ATSAMD21G18A
Chip ID      : 10010005
Version      : v2.0 [Arduino:XYZ] May  9 2016 16:06:46
Address      : 8192
Pages        : 3968
Page Size    : 64 bytes
Total Size   : 248KB
Planes       : 1
Lock Regions : 16
Locked       : none
Security     : false
Boot Flash   : true
BOD          : true
BOR          : true
Arduino      : FAST_CHIP_ERASE
Arduino      : FAST_MULTI_PAGE_WRITE
Arduino      : CAN_CHECKSUM_MEMORY_BUFFER
Erase flash
done in 0.664 seconds

Write 8608 bytes to flash (135 pages)

[==============                ] 47% (64/135 pages)
[============================  ] 94% (128/135 pages)
[==============================] 100% (135/135 pages)
done in 0.071 seconds

Verify 8608 bytes of flash with checksum.
Verify successful
done in 0.007 seconds
CPU reset.
```

whereas in platformio:
```
SAM-BA operation failed
Send auto-baud
Set binary mode
readWord(addr=0)=0x20007ffc

readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010305
version()=v2.0 [Arduino:XYZ] May  9 2016 16:06:46
chipId=0x10010005
Connected at 115200 baud
readWord(addr=0)=0x20007ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010305
Atmel SMART device 0x10010005 found
write(addr=0x20004000,size=0x34)
*** [upload] Error 1
========================== [ERROR] Took 16.14 seconds ==========================
```

with:
```ini
[env:sodaq_one]
platform = atmelsam
board = sodaq_one
framework = arduino
```

I can get it to upload with a custom board: https://github.com/joscha/platform-sodaqsamd/blob/platformio/boards/sodaq_oneUSB.json definition which is slightly different to https://github.com/platformio/platform-atmelsam/blob/master/boards/sodaq_one.json (maybe @ivankravets can comment as to the differences?)
I had created https://github.com/joscha/framework-sodaqsamd and https://github.com/joscha/platform-sodaqsamd before the SODAQ boards were integrated into pio as a first-class citizen and would love to retire those forks.

References https://github.com/platformio/platformio-core/issues/796

@Roman3349 you had any luck uploading a sketch via `pio` to any of the boards? Could you share the config?

@aethaniel, do you have any idea as you created most of https://github.com/SodaqMoja/SodaqCore-samd - any help would be appreciated!

joscha, 2017-03-05:
It could potentially also be the size of the bootloader: https://github.com/platformio/platform-atmelsam/compare/develop...joscha:platformio#diff-2f0a21be6a8aa1aab6e8e332799f2f95R28 - I just updated the framework to 1.6.12 and I am still able to upload with my fork:

```console
platformio platform install https://github.com/joscha/platform-sodaqsamd.git
```

```ini
[env:sodaq_oneUSB]
platform = sodaqsamd
framework = arduino
board = sodaq_oneUSB
```

aethaniel, 2017-03-05:
Hi,

I contributed to Arduino samd which is certainly the source of Sodaq.

Anyway, platform may be switched to atmelsamd instead of atmelsam.
I don't have much knowledge of both platformio or sodaq...

[env:sodaq_one]
-->platform = atmelsam
board = sodaq_one
framework = arduino

ivankravets, 2017-03-05:
See source of https://github.com/platformio/platformio-pkg-framework-arduinosam

> Anyway, platform may be switched to atmelsamd instead of atmelsam.

Our `atmelsam` development platform combine all family if SAM chips.

joscha, 2017-03-07:
@ivankravets any idea why my fork works and the `platform-atmelsam` doesn't?

ivankravets, 2017-03-07:
@joscha I'll have a look at it after PlatformIO IDE 2.0 release. Please sorry.

P.S: Don't forget to ping me here after PIO IDE release.

joscha, 2017-03-29:
@ivankravets ping and congratulations to the 3.3.0 release!

ivankravets, 2017-03-30:
@joscha could you try in your `platformio.ini` 1.5.0 version of this dev platform?

```
[env:...]
platform = atmelsam@1.5.0
```

See PIO IDE 2.0 
[image]

ivankravets, 2017-04-06:
Please re-test current development branch (2.0.0-alpha.1):

* Uninstall existing development platform `pio platform uninstall atmelsam`
* Install development version `pio platform install https://github.com/platformio/platform-atmelsam.git`

vicsusq, 2017-04-07:
New to PlatformIo and having exactly the same problem uploading to a Sodaq Explorer board. Tried the current development branch without success:

BeforeUpload(["upload"], [".pioenvs\sodaq_explorer\firmware.bin"])
Auto-detected: COM65
Forcing reset using 1200bps open/close on port COM65
Waiting for the new upload port...
"bossac" --info --debug --port "COM65" --erase --write --verify --reset -U false .pioenvs\sodaq_explorer\firmware.bin
Send auto-baud

SAM-BA operation failedSet binary mode

readWord(addr=0)=0x20007ffc
readWord(addr=0xe000ed00)=0x410cc601
r*eadWord(addr=0x41002018)=0x10010300** [upload] Error 1

version()=v2.0 [Arduino:XYZ] May  9 2016 16:06:46
chipId=0x10010000
Connected at 115200 baud
readWord(addr=0)=0x20007ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010300
Atmel SMART device 0x10010000 found
write(addr=0x20004000,size=0x34)
=================================== [ERROR] Took 25.51 seconds ===================================

ivankravets, 2017-04-07:
1. Please share on http://pastebin.com a full output of build+upload process from Arduino IDE. You can enable it via `Arduino IDE > Settings > Verbose mode for build/upload`
2. Process project with PIO in verbose mode via `pio run -t upload -v` and share these results on http://pastebin.com too.

We will compare both outputs and check what is wrong. Thanks!

vicsusq, 2017-04-09:
Sorry for the delay - been away for a couple of days. Both outputs are here:
[https://pastebin.com/F0epbuYA]
PlatformIo output seems to be interleaved messages from two threads.

vicsusq, 2017-04-10:
The bossac command from the Arduino IDE has "-U true" while the PlatformIo bossac command has "-U false". Running bossac directly from the command line with "-U true" will upload fine. Running with "-U false" fails every time.

closed 2017-04-10 by ivankravets

vicsusq, 2017-04-10:
Working fine now with Sodaq Explorer board thank you.

joscha, 2017-04-11:
\o/ seems to work for me as well, except for a few glitches with the 2.0.0 beta (see https://github.com/platformio/platformio-atom-ide/issues/356)

Thanks a lot @ivankravets!

VioletGiraffe, 2018-02-23:
Same problem and the same error with Arduino Due and the current PlatformIO.
The same flag `-U false` is set by PIO.

commit f8abe991 (2017-04-10): Fix uploading to USB-native boards using SAMBA // Resolve #10 Resolve #11