platformio/platform-atmelsam #11: No load on Arduino MkrZero

PLambertk, 2017-02-23:
Unable to load the basic Blink program to an Arduino MkrZero, Compile completes OK, then  I get the following messages :

Looking for upload port…
Auto-detected: COM6
Forcing reset using 1200bps open/close on port COM6
Uploading .pioenvs\mkrzero\firmware.bin
Atmel SMART device 0x100110005 found
SAM-BA operation failed
*** [upload] error 1
[ERROR] took 46.3 seconds

ivankravets, 2017-04-06:
Please re-test current development branch:

* Uninstall existing development platform `pio platform uninstall atmelsam`
* Install development version `pio platform install https://github.com/platformio/platform-atmelsam.git`

PLambertk, 2017-04-07:
Thank you Ivan for taking the time to address the issue.
My project being rather urgent I have switched to a different IDE since I reported the issue. However I will get back to platformio when I'm done and will report my test results then.

ivankravets, 2017-04-07:
Would be thankful if you help us to resolve this issue!

ivankravets, 2018-02-03:
If someone experience this issue, please provide more details

commit f8abe991 (2017-04-10): Fix uploading to USB-native boards using SAMBA // Resolve #10 Resolve #11
closed 2018-10-10 by ivankravets