platformio/platform-atmelsam #12: Blink example doesn't run on Arduino Zero after upload

chrise96, 2017-03-02:
In eclipse I imported a Platformio project of the `zero`.
Initialize platformio project using: `platformio init --ide eclipse --board zero`

`mzeropro` is working on my .org arduino m0 pro. (Bootloader_D21)
But the `zero` doesn't work for me with the .cc Arduino Zero. (Bootloader SAM-BA)

After I upload a simple blink sketch, my board looks dead and after this programming in the Arduino IDE doesn't work either... No errors are shown... But no leds toggle.
I have to burn the bootloader again in the Arduino IDE, and the board can be used again.

In the console everything looks fine (blink.cpp in Platformio zero project):
```
Info : at91samd21g18.cpu: hardware has 4 breakpoints, 2 watchpoints
target state: halted
target halted due to debug-request, current mode: Thread
xPSR: 0x41000000 pc: 0x00000614 msp: 0x20007ffc
** Programming Started **
auto erase enabled
Info : SAMD MCU: SAMD21G18A (256KB Flash, 32KB RAM)
wrote 16384 bytes from file .pioenvs/zero/firmware.bin in 1.925072s (8.311 KiB/s)
** Programming Finished **
** Verify Started **
verified 9760 bytes in 0.823946s (11.568 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked
```

chrise96, 2017-03-07:
PlatformIO: Upload session erases the bootloader. Any help?

chrise96, 2017-03-17:
Still no luck with debugging the Arduino zero. So, I made a video to clarify the issue.

https://youtu.be/VsR1WFR9vbw

Bootloader from the Arduino M0 pro is working (0:04)
Arduino Zero (sam-ba) debugging session erases bootloader (0:20)
Using the latest OpenOCD github build (1:27)

chrise96, 2017-03-18:
@ivankravets

Still no luck with debugging the Arduino zero. So, I made a video to clarify the issue.

https://youtu.be/VsR1WFR9vbw

Bootloader from the Arduino M0 pro is working (0:04)
Arduino Zero (sam-ba) debugging session erases bootloader (0:20)
Using the latest OpenOCD github build (1:27)

PatrikSB, 2017-04-03:
I have the same problem as described above.
Bootloader needs to be re-burned from arduino IDE for the board to work again.

ivankravets, 2017-04-03:
It's already fixed on my local machine. I'll deploy fix this week. We are working on the very interesting feature for Arduino Zero/M0 Pro :)

PatrikSB, 2017-04-03:
Cool! :) Was it a big fix? Possible to describe it, so I can get the environment up and running sooner? :P

ivankravets, 2017-04-03:
Ok, will try to deploy changes today :)

PatrikSB, 2017-04-03:
Thank you so much! :D

ivankravets, 2017-04-06:
Please re-test current development branch:

* Uninstall existing development platform `pio platform uninstall atmelsam`
* Install development version `pio platform install https://github.com/platformio/platform-atmelsam.git`

chrise96, 2017-04-07:
Followed the steps you mentioned and initialized a new project. Sadly, the fix is not working. Output of the last lines uploading in verbose mode: 

```
Error: at91samd21g18.cpu -- clearing lockup after double fault
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x61000003 pc: 0xfffffffe msp: 0xffffffd8
shutdown command invoked
```

So, after uploading, the bootloader will be wiped. The Arduino Zero .cc version is using the SAM-BA bootloader (for USB and UART interfaces). To use SAM-BA together with an application, we need to link the application starting at 0x2000.

Tried this in boards.txt, location SAMD package. But, no luck:
> arduino_zero_edbg.bootloader.size=0x2000

ivankravets, 2017-04-07:
1. Please note that we have 2 different configuration profiels for Arduino Zero:
   * `zero` - you have to use Programming/Debug port 
   * `zeroUSB` - you have to use USB port to program a board
2. Please share on http://pastebin.com a full output of build+upload process from Arduino IDE. You can enable it via `Arduino IDE > Settings > Verbose mode for build/upload`
3. Process project with PIO in vermose mode via `pio run -t upload -v` and share these results on http://pastebin.com too.

We will compare both outputs and check what is wrong. Thanks!

chrise96, 2017-04-07:
Arduino IDE:
https://pastebin.com/DSqLhPX1

PlatformIO:
https://pastebin.com/5m3BnmLr

ivankravets, 2017-04-07:
Could you run PIO with `--verbose` flag?

If you use PIO IDE, please open `src/blink.ino`, then PIo IDE Terminal and run
```
pio run -t upload -v
```

chrise96, 2017-04-07:
platformio run -e zero -t upload -v
https://pastebin.com/qtrMcJsr

ivankravets, 2017-04-07:
Could you try again to flash board using Arduino IDE and then PIO? 

Take a look at 22 line [here](https://pastebin.com/qtrMcJsr). We don't erase your bootloader now.

chrise96, 2017-04-10:
pio run -t upload -v
https://pastebin.com/3zkv66bh

chrise96, 2017-04-26:
It must be something with the build target inside Eclipse... I copied the complete SAMD package version 1.6.14 from Arduino in the right locations of ~/.platformio/packages/framework-arduinosam and still the same issue

perception-aweichman, 2017-05-24:
Any resolution to this?  I'm having this issue as well.  I can support debugging if anyone is still available to help resolve this.

chrise96, 2017-05-24:
No, I haven't fixed it with PlatformIO so used the Arduino IDE build and upload paths in Eclipse: https://www.youtube.com/watch?v=FRecvWhprhs&t=30s

perception-aweichman, 2017-05-24:
So after digging more, I think this is a change in the OpenOCD.  I had this process working back in December of 2016, but it looks like there was a new version of openocd in January.  The root cause of this seems to be that openocd is erasing the 0-0x1fff block of flash in addition to the 0x2000 and above area where the program is expecting to be placed.  We may be able to solve this with a linker command to place the compiled code at address 0.  I'll keep digging into how we can fix this.

ivankravets, 2017-05-24:
Could someone provide VERBOSE output of 
```
pio run --verbose -t upload
```

Thanks!

perception-aweichman, 2017-05-24:
sure ivankravets.  Do you need the whole run or just the last few lines?  Here are the relevant compile and flash lines:
https://pastebin.com/mQgYFpEL

Note the following:
line 4: **verify reset 0x2000; shutdown"**
line 29: **Warn : Adding extra erase range, 00000000 to 0x00001fff**

perception-aweichman, 2017-05-24:
Also, I just confirmed that if I program with PIO, then reprogram the bootloader with the arduino IDE, the program works as expected.

ivankravets, 2017-05-25:
1. Please restore bootloader with Arduino IDE
2. Try to edit `~/.platformio/packages/framework-arduinosam/variants/arduino_zero/openocd_scripts/arduino_zero.cfg`
```
source [find interface/cmsis-dap.cfg]

# chip name
set CHIPNAME at91samd21g18
set ENDIAN little

# choose a port here
set telnet_port 0

at91samd bootloader 8192

source [find target/at91samdXX.cfg]
```
3. Upload firmware with PIO.

Does it work?

perception-aweichman, 2017-05-26:
openocd doesn't seem to recognize the at91samd command:

GNU ARM Eclipse 64-bits Open On-Chip Debugger 0.10.0-00114-g84195369 (2017-04-19-00:49)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
/Users/user/.platformio/packages/framework-arduinosam/variants/arduino_zero/openocd_scripts/arduino_zero.cfg:30: Error: invalid command name "at91samd"
in procedure 'script'
at file "embedded:startup.tcl", line 60
at file "/Users/user/.platformio/packages/framework-arduinosam/variants/arduino_zero/openocd_scripts/arduino_zero.cfg", line 30
*** [upload] Error 1

Also on pastebin:
https://pastebin.com/caqKB5Py

valeros, 2017-05-26:
Hi @perception-aweichman !
Could you please try to restore your board to the default state (with bootloader) and then upload the blink sketch with this environment:
```ini
[env:zero]
platform = atmelsam
board = zero
framework = arduino
upload_flags = -d3 --log_output log.txt
```
And after programming please upload log.txt to pastebin.com. Thanks!

perception-aweichman, 2017-06-01:
Sorry for the delay.  Here is the log (I removed some PII from it before uploading).

It appears to be too large for pastebin...  I'll just attach it here instead....
[log.txt](https://github.com/platformio/platform-atmelsam/files/1045723/log.txt)

perception-aweichman, 2017-06-02:
Also, as a side note, I have a MKRZero (which is based on the same ATMELSAM framework) that doesn't have this same issue.  It flashes with PIO and runs just fine afterward.  It may not have the same bootloader structure as the Zero though.

perception-aweichman, 2017-06-12:
Any progress on this?  Anything I can do to assist the debugging?

ivankravets, 2017-06-13:
@perception-aweichman please mail me@ikravets.com for further details

ivankravets, 2017-06-26:
Please update development platform to v2.2.0 (`pio update`). Does it work now?

ivankravets, 2017-07-20:
Please create a new issue if you have a problem.

closed 2017-07-20 by ivankravets

ivankravets, 2017-09-07:
Finally! We have fixed all issues with Arduino Zero. Please update to 2.6.0:

- https://github.com/platformio/platform-atmelsam/releases/tag/v2.6.0

ivankravets, 2018-02-10:
Guys, we have made a huge update to Atmel SAM dev/platform. Integrated popular debuggers, custom upload tools/protocols, etc.

Could re-test the latest upstream version with Arduino Zero?

* http://docs.platformio.org/en/latest/platforms/atmelsam.html#stable-and-upstream-versions

We don't have this board and see that it is very specific. We updated openOCD to 0.10.0 and maybe it will not work with Arduino Zero. Please try both ports: USB Native and Programming.

Thanks!

/cc @chrise96 @perception-aweichman @PatrikSB

commit 288c1c31 (2017-04-06): PIO Debug; Fix uploading to Arduino Zero; Update SAMD core to 1.6.14; New board Arduino MKRFox1200 // Resolve #12