platformio/platform-atmelsam #13: PIO doesn't wait long enough for port open/close (SAMD21G)

SRGDamia1, 2017-05-16:
What kind of issue is this?

    [*] PlatformIO Core. If you’ve found a bug, please provide an information below.

You can erase any parts of this template not applicable to your Issue.
Configuration

Operating system: Windows 7 SP1

PlatformIO Version (platformio --version): v3.4.0b2 (within Atom: 2.0.0b.3)
Description of problem

When attempting to upload to an SAMD21 board (Adafruit Feather M0), the upload force closes the port, but does not wait long enough for Windows to actually close and re-open the serial port and so the upload fails. Sometimes Windows is speedy enough with the ports, but other times it misses.
Steps to Reproduce

    Create sketch for Adafruit Feather M0
    Upload using standard upload button in Atom (pio run --target upload)

Actual Results

Sometimes works without error
Other times, the port is forced closed, but before Windows has close the port, the uploader detects the old port as if it was the new port and attempts to upload to just as it is closing, causing a failure
Expected Results

Successful upload
Additional info

The Arduino IDE does not seem to have trouble with the port open/close and the speed of the port open/close on my system seems to depend some on what code the M0 had been running.

ivankravets, 2018-02-10:
Could you reproduce this issue with the latest upstream version? 

* http://docs.platformio.org/en/latest/platforms/atmelsam.html#stable-and-upstream-versions

ivankravets, 2018-03-13:
Please reopen if you still need help

jlillest, 2018-10-16:
This seems to be a problem in the 1200baud sync.  Can someone point me in the right direction for making changes to test?

ivankravets, 2018-10-16:
How did you install PIO Core? You can try to tune [WaitForNewSerialPort)(https://github.com/platformio/platformio-core/blob/develop/platformio/builder/tools/pioupload.py#L56).

hansdereber, 2018-11-11:
I experience a problem similar to the one reported by @jlillest . But for me it never works. Also random resetting and un- and re-plugging does not help.

I am running PlatformIO from the VSCode in version 3.6.2a2.

When trying to upload a minimal sample to the adafruit_feather_m0 board I get the following error:
````
Configuring upload protocol...
AVAILABLE: atmel-ice, blackmagic, jlink, sam-ba
CURRENT: upload_protocol = sam-ba
Looking for upload port...
Auto-detected: /dev/ttyUSB0
Forcing reset using 1200bps open/close on port /dev/ttyUSB0
Waiting for the new upload port...
Uploading .pioenvs/adafruit_feather_m0/firmware.bin
No device found on ttyUSB0
*** [upload] Error 1
````

For now I am resorting to the ArduinoIDE, with which the upload to the board works without problems.

Any help will be highly apreciated.