platformio/platform-atmelsam #26: Unable to upload to samd21_xpro

sampscl, 2018-01-30:
I am unable to program my Atmel SAM D21 Xplained Pro. I have run `pip install -U platformio` and `pip install -U https://github.com/platformio/platformio-core/archive/develop.zip` on Ubuntu 16.04 and had the same results (ignoring the "Obsolete PIO Core warning...):
```bash
platformio run -t upload -v
**************************************************************************************************************************
Obsolete PIO Core v3.5.1 is used (previous was 3.5.2a6)
Please remove multiple PIO Cores from a system:
http://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
**************************************************************************************************************************
[Tue Jan 30 09:42:01 2018] Processing samd21_xpro (platform: atmelsam; board: samd21_xpro; framework: mbed)
--------------------------------------------------------------------------------------------------------------------------
Collected 22 compatible libraries
Scanning dependencies...
No dependencies
"openocd" -s /home/user/.platformio/packages/tool-openocd/scripts -f atmel_samd21_xplained_pro.cfg -c "telnet_port disabled; program {{.pioenvs/samd21_xpro/firmware.bin}} verify reset ; shutdown"
Open On-Chip Debugger 0.9.0-g932ec70 (2017-02-15-11:32)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
Runtime Error: embedded:startup.tcl:60: Can't find atmel_samd21_xplained_pro.cfg
in procedure 'script'
at file "embedded:startup.tcl", line 60
*** [upload] Error 1
=============================================== [ERROR] Took 1.14 seconds ===============================================
```

It seems that openocd is being told to look for `atmel_samd21_xplained_pro.cfg` in the wrong place. That config file is in `/home/user/.platformio/packages/tool-openocd/scripts/board/atmel_samd21_xplained_pro.cfg`, not ` /home/user/.platformio/packages/tool-openocd/scripts`.

sampscl, 2018-02-05:
An update: this command worked from the terminal:
```
/home/user/.platformio/packages/tool-openocd/bin/openocd -s /home/user/.platformio/packages/tool-openocd/scripts -f board/atmel_samd21_xplained_pro.cfg -c "telnet_port disabled ; program .pioenvs/samd21_xpro/firmware.bin verify reset ; shutdown"

Open On-Chip Debugger 0.9.0-g932ec70 (2017-02-15-11:32)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
adapter speed: 500 kHz
adapter_nsrst_delay: 100
cortex_m reset_config sysresetreq
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 01.1A.00FB
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 500 kHz
Info : SWD IDCODE 0x0bc11477
Info : at91samd21j18.cpu: hardware has 4 breakpoints, 2 watchpoints
target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x21000000 pc: 0x000016a8 msp: 0x20002168
** Programming Started **
auto erase enabled
Info : SAMD: partitioning the first flash page into 8 subpages
Info : SAMD MCU: SAMD21J18A (256KB Flash, 32KB RAM)
wrote 2048 bytes from file .pioenvs/samd21_xpro/firmware.bin in 0.243147s (8.225 KiB/s)
** Programming Finished **
** Verify Started **
verified 1048 bytes in 0.122116s (8.381 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked
```

ivankravets, 2018-02-10:
Please re-test with the latest upstream (GIT) version http://docs.platformio.org/en/latest/platforms/atmelsam.html#stable-and-upstream-versions

closed 2018-02-10 by ivankravets

sampscl, 2018-02-12:
Yes, that has fixed my problem.

commit c09166e1 (2018-02-10): Add support for J-Link, Atmel Ice, BlackMagic tools for debugging and uploading // Resolve #18 Resolve #26