platformio/platform-atmelsam #27: Industruino support

vshymanskyy, 2018-02-09:
This should be rather easy, but it has some pitfalls. First, it uses modified bossac-0x4000 tool (different bootloader size). Should be easier with this sorted out:
https://github.com/shumatech/BOSSA/issues/33
Also, needs to be added to https://github.com/platformio/platformio-core/blob/develop/scripts/99-platformio-udev.rules
```
# Industruino
ATTRS{idVendor}=="2e78", ATTRS{idProduct}=="004d", ENV{ID_MM_DEVICE_IGNORE}="1"
ATTRS{idVendor}=="2e78", ATTRS{idProduct}=="004d", ENV{MTP_NO_PROBE}="1"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="2e78", ATTRS{idProduct}=="004d", MODE:="0666"
KERNEL=="ttyACM*", ATTRS{idVendor}=="2e78", ATTRS{idProduct}=="004d", MODE:="0666"
```

SebDominguez, 2019-01-19:
The Industruino board is the only reason that I still use the Arduino IDE.

albydnc, 2019-05-28:
is it resolved?

cheetor5923, 2019-08-12:
Nope, nobody seems to care. this has been classified (incorrectly) as an enhancement instead of a bug.

cheetor5923, 2019-08-13:
I'm now actively attempting to fix this problem

kumajaya, 2021-04-15:
1. Install PlatformIO IDE on top of Microsoft Visual Studio Code:
https://docs.platformio.org/en/latest/integration/ide/vscode.html#installation

2. In PlatformIO Home - Platforms - Advanced Installation, paste the following repo (industruino branch):
https://github.com/kumajaya/platform-atmelsam#industruino

3. Back to https://docs.platformio.org/en/latest/integration/ide/vscode.html#quick-start, Industruino D21G and Industruino 4-20mA.ker boards will be available for you in PlatfomrIO IDE board selection menu.

Upload to your board at your own risk! No Industruino board near me for now.