platformio/platform-atmelsam #30: Unable to upload to saml21_xpro_b

pga23, 2018-04-25:
After compilation and upload I have the following:

```
Linking .pioenvs/saml21_xpro_b/firmware.elf
Checking program size
text       data     bss     dec     hex filename
1028         36   32584   33648    8370 .pioenvs/saml21_xpro_b/firmware.elf
Building .pioenvs/saml21_xpro_b/firmware.bin
Configuring upload protocol...
AVAILABLE: atmel-ice, blackmagic, cmsis-dap, jlink
CURRENT: upload_protocol = cmsis-dap
Uploading .pioenvs/saml21_xpro_b/firmware.bin
GNU MCU Eclipse 64-bits Open On-Chip Debugger 0.10.0+dev-00392-gbe9ef0b0 (2018-01-12-14:56)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
little
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
adapter speed: 500 kHz
adapter_nsrst_delay: 100
cortex_m reset_config sysresetreq
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 03.25.01B6
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 500 kHz
Info : SWD DPIDR 0x0bc11477
Info : at91saml21j18.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x00000854 msp: 0x20002048
** Programming Started **
auto erase enabled
Info : SWD DPIDR 0x0bc11477
Error: Failed to read memory at 0x400e0420
Error: Cannot read SAM3 register: CKGR_MOR @ 0x400e0420, Err: -4
Error: Cannot read SAM3 register: CKGR_MOR @ 0x400e0420, Error: -4
Error: auto_probe failed
embedded:startup.tcl:476: Error: ** Programming Failed **
in procedure 'program'
in procedure 'program_error' called at file "embedded:startup.tcl", line 532
at file "embedded:startup.tcl", line 476
*** [upload] Error 1
```

My platformio.ini is:
```
[env:saml21_xpro_b]
platform = atmelsam
framework = mbed
board = saml21_xpro_b
```

I'm using Ubuntu 14.04 and PlatformIO: Home 0.9.5 Core 3.5.3b1

Brand new SAML21 board, on AtmelStudio7 (Windows 7) upload works perfectly.

ivankravets, 2018-04-25:
Please re-test with upstream version http://docs.platformio.org/en/latest/platforms/atmelsam.html#stable-and-upstream-versions

Does it work now?

pga23, 2018-04-25:
It is working now - thanks a lot for the fix!

commit 2e0cea6c (2018-05-02): Fix incorrect debug target for SAML21 // Resolve #30