platformio/platform-atmelsam #31: Add support for debut via STLink

augustozanellato, 2018-05-20:
Can you please add support for debugging Atmel SAM (specifically ATSAM3X8E (Arduino Due)) via STLink and OpenOCD?

ivankravets, 2018-10-10:
Is it possible?

augustozanellato, 2018-10-10:
Yes, if you use the StLink via openocd you can debug whatever mcu you want if it has jtag or swd support by openocd

ivankravets, 2018-10-10:
Have you tried this? Does it work for you?

augustozanellato, 2018-10-10:
I tried this when I opened the issue, I can connect to the atsam3x8e using gdb via openocd and it works with some slight modifications to openocd config files

ivankravets, 2018-10-10:
Great! Could you provide your configuration and changes? I would be glad to test it. I have ST-Link and Due board.

augustozanellato, 2018-10-10:
I have to check if I still have the files

augustozanellato, 2018-10-10:
I lost the files when I wiped my pc this summer because they were in ~/.platoformio and I don’t have backed up hidden folders, in the next days I’m going to try to get it to work again

ivankravets, 2018-10-10:
Good, please ping me here.

augustozanellato, 2018-10-11:
https://gist.github.com/augustozanellato/961dec48b881330dfd5a7ac34aa1e268
Here you are, I didn't try it extensively, but gdb connects and I can halt/run/reset the cpu, so the swd communications works fine

ivankravets, 2018-10-12:
Thank you so much!!! A few generic hacks for all people and it now works with 1-click with PlatformIO :) 
Please note that we use "preload" mode to flash debugging firmware. See https://electronics.stackexchange.com/questions/250664/can-i-use-st-link-programmer-for-non-st-chips

The right configuration is 
```
[board:due]
...
debug_tool = stlink
```

Also, please use programming port.

[image]

closed 2018-10-12 by ivankravets

augustozanellato, 2018-10-12:
I think that PlatformIO now is the first development environment to support debugging Atmel SAM with ST-Link out of the box

ivankravets, 2018-10-12:
Could you try upstream version http://docs.platformio.org/en/latest/platforms/atmelsam.html#upstream? Does it work for you?

augustozanellato, 2018-10-12:
Ok, I can confirm it work, the connection is even stabler than with my jlink v8

ivankravets, 2018-10-12:
I also like it :) See https://twitter.com/PlatformIO_Org/status/1050738608070713344

Happy coding with PlatformIO! ;)

augustozanellato, 2018-10-12:
You got a like from Microchip Makes, but I don't think you are going to get one from ST :D

commit b19adfd8 (2018-10-12): Add support for debug via ST-Link // Resolve #31