platformio/platform-atmelsam #33: Atmel ASF3 as alternative framework?

Timvrakas, 2018-07-27:
I'm interested in using PlatformIO to develop for the SAMD21 series, using the Atmel Software Framework. To achieve this, I need to set up ASF as a _framework_ correct? Like framework-arduinosam, but it would be "framework-asf" or something like that.

Where should I look to get an idea of how to achieve this? Is there any relevant documentation? For the short term i'm just looking to port SAM21G18A, but I could expand that to more of the ASF later on.

I'm new to PIO, so any help would be greatly appreciated!

Timvrakas, 2018-09-11:
I'm back...

Timvrakas, 2018-09-11:
I'm back...
I put together an ASF3 based demo program in Atmel Studio, then I compiled it with the atmel-provided makefile. I setup an PIO environment with the AtmelSam Platform, and my board (the SFE Mini), but with no framework in use. by adding -I includes manually in platformio.ini, I was able to get the code working.

The one thing I had to change was the GCC version. It would not compile with 1.40804. I knew it worked with the lastest `gcc-arm-embedded` on ubuntu, So I upgraded to 1.70201 in PIO and then it worked.
I have a few questions:

1. What are those version numbers from? are the 1.xxx versions of arm-none-eabi PIO specific, within the PIO manager?
2. the atmelsam platform specifies using 1.40804, but I seem to need 1.70201. If I add a framework, can I override the toolchain version somehow?

Thanks
Tim

Timvrakas, 2018-09-11:
In addition, many of the modules within ASF have user configuration header files. for example, when you install FreeRTOS-10.0.0 with ASFWizard in Atmel Studio, `module_config/FreeRTOSConfig.h` gets copied into the user project. Is there a similar way to do that in PIO?

jeffsf, 2018-11-17:
I'm definitely interested in helping out on this as I can, as getting off Arduino is high on my list right now. 

If you have the order of library search for the SAMD21 from Atmel Studio, that would be greatly appreciated. No Windows machines here, unfortunately.

Timvrakas, 2018-11-17:
Yes, here is a list of all the files I had to include for ASF3: https://github.com/Timvrakas/Code_Demo/blob/pio/lib/ASF3/library.json

However, I didn't realize that there was a ASF4 available, and it's far superior in my opinion. If a get a chance, I'm going to try to get that working. 

The real cool feature of asf-pio integration would be code generation similar to Atmel Start. Pull pin names/pin modes/mux/timer/sercom configuration from a JSON file or something, and generate all the headers at compile time. But honestly that large of a project would probably only be doable with support from Microchip.

But basic compiling is doable in the short term.

jeffsf, 2018-11-17:
Thanks! Looking at ASF4 now -- that Start tool is painfully slow. I did see that [Adafruit has some extracts](https://github.com/adafruit/asf4), but they tend to only pull what they need to support their own boards and code.

Timvrakas, 2019-02-06:
I did get this working. My code is here: https://github.com/stanford-ssi/spacesalmon

frankleonrose, 2019-10-11:
I'm considering working on an ASF framework. I have some questions and I'd appreciate guidance, @valeros.

1. Looks like it would have to be ASF 3.47, since ASF 4 comes only with the Atmel tools (not available as a separate download, see https://www.microchip.com/mplab/avr-support/advanced-software-framework).

2. Since the license isn't totally permissive, I'm leery of committing the entire ASF directory to the framework repo. Is there precedent for a framework package to download a resource (asf-standalone-archive-3.47.0.96.zip) during installation? Is there a hook for that or would it have to be done by the builder?

3. Different boards will require different files in from the ASF directory tree, therefore the builder will have to add source filters to select files to build and files to ignore. Similarly, the builder will need to add to the include path based on mcu/board. Is there a better way of doing that?

4. ASF out of the box supports a relatively small number of Microchip dev boards (samXYZ_xplained_pro). I imagine having a variants directory with board-specific subdirs like variants/adafruit_feather_m0 containing initialization source, configuration headers, and linker scripts if necessary. I could add the boards I use and expect pull requests to add others.

5. If, for example, I added the Adafruit Feather M0, in order to build it with `framework = asf` the board.json file would have to include "asf" in the frameworks section. I'd have to submit a PR to make that change, right?

Thanks for your help.

valeros, 2019-10-11:
Hi @frankleonrose ! 
> Looks like it would have to be ASF 3.47, since ASF 4 comes only with the Atmel tools (not available as a separate download, see https://www.microchip.com/mplab/avr-support/advanced-software-framework).

I'm not familiar with ASF ecosystem, but if it's possible to compile this framework using GCC toolchain then we're mainly interested in sources and build flags that can be possibly used for compilation. According to the ASF doc page, ASF3 indeed looks like a better choice.

> Since the license isn't totally permissive, I'm leery of committing the entire ASF directory to the framework repo. Is there precedent for a framework package to download a resource [...]

In any case we need to repack framework because we rely on a special utility manifest with package metadata (name, version, url, etc).

> Different boards will require different files in from the ASF directory tree, therefore the builder will have to add source filters to select files to build and files to ignore. Similarly, the builder will need to add to the include path based on mcu/board. Is there a better way of doing that?

Pretty much all frameworks require some build logic, files filtering is not a problem as long as the structure of the framework is consistent.  You can take a look at [the build script for stm32cube](https://github.com/platformio/platform-ststm32/blob/develop/builder/frameworks/stm32cube.py) that might have some answers.

> ASF out of the box supports a relatively small number of Microchip dev boards (samXYZ_xplained_pro). I imagine having a variants directory with board-specific subdirs like [...]

I've never used ASF and not familiar with the typical workflow, but it should be fine to create a `variant` folder for this purpose, at least at an early stage. Eventually, it might be a good idea to dynamically generate these files.

> If, for example, I added the Adafruit Feather M0, in order to build it with framework = asf the board.json file would have to include "asf" in the frameworks section. I'd have to submit a PR to make that change, right?

That's correct.

frankleonrose, 2019-10-15:
Thanks, @valeros.

> Eventually, it might be a good idea to dynamically generate these files.

I assume you mean to generate these files while re-packaging the framework, as opposed to after installation. (Since there is no post-install hook.)

Also, copying the platform package (with platform.json and platform.py files and everything) to $PLATFORMIO_HOME/platforms/asf is sufficient to install it for testing?

Timvrakas, 2019-10-15:
I'm using ASF4 with PlatformIO for this project.
https://github.com/stanford-ssi/SpaceSalmon

ASF4 is better than ASF3, but is not very good. More exposed than Arduino
to be sure, but some broken libraries (SPI async) and lots of pointless
code. I question the value of having it built in as a PlatformIO framework
beacuse anyone who's trying to use it should be expecting to edit it for
their needs. It's a good starting point, but I just have it included as a
local library.

The end game for this chip family IMO is a Python configuration tool. Like
Atmel Start, but executed at build time, and pulling configuration from
some JSON or something. Automatic config checking to look for errors ("your
CPU has no GCLK connected!"). Then it generates a single layer framework
for exactly the hardware configuration you've specified, including
initialization code.

But that sounds like a project for someone at Microchip.

Tim

frankleonrose, 2019-10-21:
> I question the value of having it built in as a PlatformIO framework
beacuse anyone who's trying to use it should be expecting to edit it for
their needs.

That's an excellent point, @Timvrakas. Having now spent more time with the idea, I agree.

Here's how I'd like to use ASF4. You put three files in a lib directory:
- MyProject.atstart - Start project configuration file, straight from https://start.atmel.com
- library.json - hooks in for Platformio (customized only with a different name in case of multi-board projects.)
- asf_library.py - Hooked in by library.json, this file downloads the project source specified by the MyProject.atstart file (caching if no change) and wires up all the source filters, include paths, and linker script. No customization of file for library/project (it's entirely parameterized by downloaded project).

You'd build with no `framework` specified, since, as you pointed out, all the CMSIS files are included in the atzip package.

JelleRoets, 2019-11-10:
Hi,
This thread looks exactly what I'm trying to achieve: for my next project I've used a ATSAMC21E18A mcu, I would like to use platformio in vscode on OSX to program, upload and debug code to the mcu (I'm using a Segger JLink as debugger) Since no existing board in plaformio uses this mcu already, it's hard to get started though.

I've used Atmel Start to configure my mcu, with the peripherals and necessary ASF4 libraries, I needed for my project. I was able to download the atstart zip package, and after installing the arm compiler, compile the project using the produced make file. I can upload the hex file using the cli tools from segger, but there it ends. I would really like to use all the goodness from platformio, to further extend and develop my project, especially since the unified debugger from platformio already supports the Segger JLink. Since I'm working on OSX, using Atmel Studio is not an option.

So could someone explain me how you can import an atmel start project and configure platformio to use the atmel config? I guess you need to define a custom board, and platformio.ini file. do you need to develop custom python files to get it compiled or can it use the make file?

frankleonrose, 2019-11-11:
@JelleRoets Check out this example and let me know how it works for you. https://github.com/frankleonrose/AtmelStart_PlatformIO/tree/master/examples/AdafruitFeatherM0 . Just run `pio run` in the example directory to try it out. The AtmelStart_PlatformIO library will automatically:
1. Repackage .atstart file specified in platformio.ini as JSON.
2. Download an .atzip file of all the source.
3. Build the include paths and sources.
4. Perform the link using the linker script included in the package.

It caches the downloaded files, naming them with the hash of the JSON used to generate them, so it doesn't have to repeat work.

To make a custom board specification, check out https://docs.platformio.org/en/latest/platforms/creating_board.html . If your board has a bootloader, make sure to specify an `upload.offset_address` value.

frankleonrose, 2019-11-11:
I realize after re-reading this thread that the builder code in https://github.com/frankleonrose/AtmelStart_PlatformIO/library.py could actually just be added here as 'builder/frameworks/atmelstart.py'.

JelleRoets, 2019-11-11:
Hi @frankleonrose,
Many thanks for your quick answer and great work. That helped me a lot.

Unfortunately I couldn't run your scripts out of the box when cloning your above repository, I had a couple of issues:
- some env variables where missing an "_" e.g. `PROJECT_WORKSPACE_DIR` instead of `PROJECTWORKSPACE_DIR` could this be caused by running an older platformio version? I'm currently running v4.1.0
- the LDSCRIPT_PATH doesn't do anything, instead I removed the "ldscript" tag from the board.json and added the atstart linker script as a separate linker flag.
- due to a bug in platformio (https://github.com/platformio/platformio-core/issues/3264) I couldn't run your script when using a custom board.

However I was able to fix these issues and make a working blink example for the arduinoZero using Atmel Start and ASF4. Please take a look at https://github.com/JelleRoets/AtmelStart_PlatformIO or as a diff: https://github.com/frankleonrose/AtmelStart_PlatformIO/compare/master...JelleRoets:master. I'd love to hear your feedback.

I've also added a readme that describes how to get started with your own custom project.
I think this might be useful in general (after fixing the remaining todo's in the readme), since it can be used for every atstart project, so maybe a worth considering adding to the main platformio branch (@ivankravets)?

ivankravets, 2019-11-12:
> PROJECT_WORKSPACE_DIR

Yes, see https://github.com/platformio/platformio-core/blob/develop/platformio/builder/main.py#L99

We recommend using PIO Core 4 Build API.

> the LDSCRIPT_PATH

Could you provide more details?

>  so maybe worth considering adding to the main platformio branch

If this is a framework, it is better to add support directly to this repository instead of a library.

frankleonrose, 2019-11-12:
Heres's the framework builder, @ivankravets: https://github.com/platformio/platform-atmelsam/pull/84 

Resolving some CI issues. (This is one place where CircleCI does a good job - easy to run jobs simply using `circleci local --execute <jobname>`. Is there a similar easy CLI with Travis?)

JelleRoets, 2019-11-12:
> > the LDSCRIPT_PATH
> Could you provide more details?

In @frankleonrose original repo https://github.com/frankleonrose/AtmelStart_PlatformIO/blob/master/library.py#L208 he set LDSCRIPT_PATH with the correct linker script file. However setting that field didn't change the linker command for me (that might be due to specific platform build scripts in the atmelsam repo, but didn't figured that out). 
I could fix it for me by adding extra flags to the LINKFLAGS https://github.com/JelleRoets/AtmelStart_PlatformIO/blob/master/library.py#L208, however this is not a perfect solution either, because if the board file also contains a ldscript in the build section it could result in 2 `-T [linkerScript]` flags.

JelleRoets, 2020-03-05:
I recently picked up this project again, but since now PIO core 4.2.0 was released our above approach for using ASF doesn't work anymore. Somehow the compile behaviour has changed and now includes an archive step of the library, which breaks the linking.

I already figured out the problem and a possible workaround as described in https://community.platformio.org/t/archive-step-breaks-output-elf-file/12115 but any help on how to automate the workaround would be much appreciated (@ivankravets / @frankleonrose ): https://github.com/JelleRoets/AtmelStart_PlatformIO/blob/master/library.py#L201

JakeSays, 2020-08-14:
I'm working on a project that uses both ATSAMC21's and ATSAME54's, and I have found microchip's frameworks maddening. Three solutions, and all of them have major issues regarding usability. Harmony 3/mplab doesn't directly support gcc - not that I'd use mplab anyway - it has to be the worst IDE out there. Harmony 3 also doesn't support SPI slave. ASF4 is a nightmare (buggy as has been mentioned) and because of the disaster that is START. And then there is ASF3 - so far it is the most usable, although it is a PITA to configure.

I like microchip's silicon, but their tooling is just a mess.

I am tempted to write a C++ framework based on a blend of harmony 3 and ASF 3.

Timvrakas, 2020-08-14:
I ended up dropping ASF4 and restructuring my pretty sizable project to use the Adafruit Arduino Core. It has its own issues, but not as many, and for advanced peripheral interfacing you're never going to escape doing some of the work yourself. If you're considering using ASF, this is a warning! The promise of vendor libraries is hollow. (Aside from the CMSIS headers are great)

JakeSays, 2020-08-14:
I'll take a look at Arduino Core. And yes very hollow!

JakeSays, 2020-08-14:
Ah. I remember looking at adafruit's core but it doesn't support the SAMC21. It probably wouldn't take much to add it though.