platformio/platform-atmelsam #38: Version 3.4.0 : my code is not working anymore

deladriere, 2018-10-12:
I can compile and upload various codes but they don’t run like usual (rotary not working anymore / I2C display not working …)
I have a second machine still on 3.2.0 and it works fine when I compile and upload the same code (shared via Dropbox)
I reverted to 3.2.0 using pio platform install atmelsam@3.2.0 and my codes run again

maybe you are making too much modification in one go ?
could you split the update like
3.2.1 : Update Adafruit Core to 1.2.3
3 2.2 : Update Arduino Core to 1.6.19

etc …
to let us find where to look ?

here is my platformio.ini

deladriere, 2018-10-12:
I found the problem ;
I am overclocking my I2C display using Wire.setClock() to the limit before the screen freeze ;
This function was known to be faulty in the past (not respecting the frequency) but it was working fine for me until today.
The frequency issue was fixed in Arduino 1.6.18 and when you updated the Platformio sam core 3.1.0 integrating the 1.6.18 I had no issue with the speed so I kept the overclocking.

I guess you did not update that function until know?
I fixed all my code by reducing the frequency to a lower value and it working again.

Another reason to ask to be very gradual in making updates : we need to absorb the new changes from various source (Arduino + Adafruit  + Platformio in my case)

ivankravets, 2018-10-12:
The package is here https://github.com/platformio/platformio-pkg-framework-arduinosam

I recommend using semantic versioning in your project. See http://docs.platformio.org/en/latest/projectconf/section_env_general.html#platform

For example, `platform = atmelsam@~3.4.0`. So, you will receive only bug fixes without new changes.

deladriere, 2018-10-13:
@ivankravets Thanks I will keep an eye on the commits in your package
Btw Is there a way to define my own variant board in the platformio.ini?

ivankravets, 2018-10-13:
See https://docs.platformio.org/en/latest/platforms/creating_board.html

closed 2018-10-13 by ivankravets