platformio/platform-atmelsam #39: SODAQ_SARA building problem: pin definitions not declared

Midnight28, 2018-10-15:
Hello!

First of all, thank you for adding SODAQ SARA board, I've waited for it a lot :)

Anyway, I have problems with building my code for SODAQ_SARA. I would like to use this example:
https://github.com/SodaqMoja/Sodaq_nbIOT/tree/master/examples/nbIOT_test_udp
With Arduino IDE it works fine, but when I try to build it with PlatformIO, I get errors like this:

`src\main.cpp:18:19: error: 'SARA_TX_ENABLE' was not declared in this scope
#define enablePin SARA_TX_ENABLE
^
src\main.cpp:70:40: note: in expansion of macro 'enablePin'
nbiot.init(MODEM_STREAM, powerPin, enablePin, SARA_R4XX_TOGGLE, cid);   // init()
^`

I know, that pin name definitions are in variant.h, and I've updated PlatformIO, so the SODAQ_SARA variant is in ..\.platformio\packages\framework-arduinosam\variants

What am I doing wrong?

ivankravets, 2018-10-23:
We use this variant. Is it outdated? https://github.com/platformio/platformio-pkg-framework-arduinosam/tree/master/variants/sodaq_sara

Midnight28, 2018-10-23:
Yes, it was. I've updated it, now it works fine, thanks!

closed 2018-10-23 by Midnight28

ivankravets, 2018-10-23:
Stop! :) Did you use outdated `atmelsam` dev/platform? Please remove `~/.platformio/packages/framework-arduinosam` folder and start building project again. Does it work now?