platformio/platform-atmelsam #40: cannot find Seeeduino LoRaWAN

lakithaomal, 2018-11-06:
Hey all,
I am trying to find Seeeduino LoRaWAN on the Board Explorer. But it seems that it is not available yet. 
Please let me know if you guys can help me?
Thank You.

samiralavi, 2019-03-18:
It is fairly the same as Arduino Zero (native USB) board. Just need to change the board name. I recommend you to read this to find HOW to do it:
https://docs.platformio.org/en/latest/platforms/creating_board.html

I will upload it when found time.

ivankravets, 2019-03-20:
Resolved in https://github.com/platformio/platform-atmelsam/commit/9f6fa1f3c9af5ac5c3c4573b585ae7cc3ce2befc

Please re-test with upstream version http://docs.platformio.org/en/latest/platforms/atmelsam.html#stable-and-upstream-versions

closed 2019-03-20 by ivankravets