platformio/platform-atmelsam #41: Support for MoteinoM0 & CurrentRanger SAMD21G boards

LowPowerLab, 2019-01-16:
Please add support for MoteinoM0 and CurrentRanger.
Latest definitions for Arduino can be found here:
https://github.com/LowPowerLab/LowPowerLab.github.io/tree/master/MoteinoCore
Thanks!

ivankravets, 2019-02-15:
Please re-test with upstream version https://docs.platformio.org/en/latest/platforms/atmelsam.html#upstream

closed 2019-02-15 by ivankravets

LowPowerLab, 2019-02-15:
Thank you, how can I do that?

ivankravets, 2019-02-15:
1. Install PlatformIO IDE => https://platformio.org/platformio-ide
2. Open PlatformIO IDE > PIO Home > Platforms > Click on "Advanced Installlation" > Paste `https://github.com/platformio/platform-atmelsam.git`
3. Close PIO Home tab and open it again, click on "New Project", search for your boards.

Does it work?

LowPowerLab, 2019-02-15:
Oh, I'm sorry, I didn't have a chance to try it out yet. I was just so busy.
I trust you that it works, you do great work, thanks for all your support 👍

commit 09ba731d (2019-02-15): Add support for LowPowerLab CurrentRanger // Resolve #41