platformio/platform-atmelsam #47: Compiling error in USB_device.h

LorenzoDeMarco, 2019-03-04:
When compiling a simple sketch in platformio with
`#include <Arduino.h>`
the following error comes up:

```
[...]\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\4.8.4\bits\basic_string.h: In member function 'int std::basic_string<_CharT, _Traits, _Alloc>::compare(const std::basic_string<_CharT, _Traits, _Alloc>&) const':
[...]\framework-arduinosam\system\sam\libsam/include/USB_device.h:364:25: error: expected unqualified-id before '(' token

#define Min(a, b)       (((a) < (b)) ?  (a) : (b))
^
```

The macro seems perfect and I never use it in my code, so what could have gone wrong?

ivankravets, 2019-03-04:
Could you provide a simple PlatformIO project to reproduce this issue?

LorenzoDeMarco, 2019-03-27:
It now compiles! Perhaps it's a bug in VS Code?

closed 2019-03-27 by LorenzoDeMarco