platformio/platform-atmelsam #57: [QUESTION] Grand station M4 project

GMagician, 2019-05-11:
I'm trying to see what does it mean to implement adafruit grand station M4 into Marlin but I'm lost with a strange error.

> AssertionError: Missed J-Link Device ID for malyanM200v2

it's generated when I execute

> platformio boards

inside marlin folder where platformio.ini is resident.

From original marlin file I only added:

> #
> # Ardfruit Metro Grand Central M4 (Atmel SAMD51P20A ARM Cortex-M4)
> #
> [env:adafruit_grandcentral_m4]
> platform          = atmelsam
> framework         = arduino
> board             = adafruit_grandcentral_m4
> build_flags       = ${common.build_flags}
> lib_deps          = ${common.lib_deps}
> src_filter        = ${common.default_src_filter} +<src/HAL/HAL_GRANDCENTRAL_M4>

this is error I get:
[errlog.txt](https://github.com/platformio/platformio-core/files/3168194/errlog.txt)

I see from closed issues that this error has been already noticed by other guys but no solution has been given

>  `pip install -U platformio` command
> platformio upgrade
> platformio update

have been executed but don't solve it

[platformio.ini.txt](https://github.com/platformio/platformio-core/files/3168205/platformio.ini.txt)

Note: It will be impossible to me to answer till next week

ivankravets, 2019-05-11:
Where did you get `malyanM200v2` board? I don't see this board in our registry.

GMagician, 2019-05-16:
> Where did you get malyanM200v2 board? I don't see this board in our registry.

This is what I get when I launch "platformio boards" it should only return supported boards, but I get reported error

ivankravets, 2019-05-17:
Which source code do you use? Where did you get it?

GMagician, 2019-05-17:
Marlin source code? or platformio code?
I don't use platformio source, I downloaded from web site the window compiled version

ivankravets, 2019-05-17:
Do you use the latest version of Marlin firmware?
- https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

I don't see this board. https://github.com/MarlinFirmware/Marlin/search?q=malyanM200v2&unscoped_q=malyanM200v2

GMagician, 2019-05-17:
Yes I just redone everything again yesterday night. Downloaded marlin 2.0.x, "update" and "upgrade" platformio and "platformio boards" and got same error.
But since "boards" should report platformio supported boards...how Marlin code may be involved?

ivankravets, 2019-05-17:
It seems that you installed another dev/platform with a broken `board` manifest. Please remove `~/.platformio` folder and restart VSCode.

ivankravets, 2019-05-17:
You can check dev/platforms via `pio platform list` or visiting `~/.platformio/platforms`. The broken `malyanM200v2.json` is located in `boards` folder of one of dev/platform.

closed 2019-05-17 by ivankravets

GMagician, 2019-05-17:
I'll try but I don't know how I could downloaded that...never done anything else that "platformio update" "platformio upgrade" and compile via VS (not vscode)

GMagician, 2019-05-17:
Thanks for support

GMagician, 2019-05-17:
I think there something else, here is "pio platform list" report:
[log.txt](https://github.com/platformio/platform-atmelsam/files/3192279/log.txt)

and C:\Users\[username]\.platformio\platforms contains only avr & saml folders

GMagician, 2019-05-17:
Text search inside files gives no result,
Get-ChildItem -r -Filter "malyan*" starting from same folder report no file

GMagician, 2019-05-17:
deleted saml folder and it works...
I executed:
  `pio init --ide visualstudio --board adafruit_grandcentral_m4`
and now I get the error again

GMagician, 2019-05-17:
next test:
deleted files and subfolder inside saml folder
now I only have
platform.json
platform.py
and I still get the same error.
error disappears only when I delete attached platform.json
[platform.json.txt](https://github.com/platformio/platform-atmelsam/files/3192433/platform.json.txt)

ivankravets, 2019-05-17:
Where did you get saml folder?

GMagician, 2019-05-17:
it is created by:
`pio init --ide visualstudio --board adafruit_grandcentral_m4`
that downloads all tool chain

GMagician, 2019-05-17:
real folder name is `atmelsam`

GMagician, 2019-05-18:
Found! Such board is define inside Marlin folders!

ivankravets, 2019-05-20:
Could you show me here https://github.com/MarlinFirmware/Marlin?

GMagician, 2019-05-20:
Of course I can, it's here:
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/buildroot/share/PlatformIO/boards
since I don't know how PlatformIO works nor I know Phyton language it has been an adventure to discover where it was the issue.
I already open an issue in Marlin repository about that
https://github.com/MarlinFirmware/Marlin/issues/14054
and guy that added board wrote:
> PIO should get proper json definitions for the M200/M300 series soon, since ST’s core now officially supports them.

ivankravets, 2019-05-20:
Resolved in https://github.com/MarlinFirmware/Marlin/pull/14070