platformio/platform-atmelsam #59: ATMEL ICE cannot upload or debug SAMD51

coolio986, 2019-05-30:
I have just tried the stable and upstream versions for the SAMD51 on windows 10. No uploading or debugging working.

> Error: error during select: Unknown error
> .pioinit:12: Error in sourced command file:
> Remote communication error. Target disconnected.: No error

> [env:adafruit_metro_m4]
> platform = atmelsam
> board = adafruit_metro_m4
> framework = arduino
> 
> debug_tool = atmel-ice
> upload_protocol = atmel-ice

ivankravets, 2019-05-30:
Could you provide full output from debug session?

coolio986, 2019-05-30:
> Reading symbols from c:\Users\Anthony\Documents\PlatformIO\Projects\ILI9341_Screen\.pio\build\adafruit_metro_m4\firmware.elf...
> 
> done.
> 
> PlatformIO Unified Debugger > http://bit.ly/pio-debug
> 
> PlatformIO: Initializing remote target...
> 
> GNU MCU Eclipse OpenOCD, 64-bitOpen On-Chip Debugger 0.10.0+dev-00593-g23ad80df4 (2019-04-22-20:25)
> 
> Licensed under GNU GPL v2
> 
> For bug reports, read
> 
> 	http://openocd.org/doc/doxygen/bugs.html
> 
> at91samd51j19
> 
> Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
> 
> none separate
> 
> adapter speed: 400 kHz
> 
> cortex_m reset_config sysresetreq
> 
> Info : tcl server disabled
> 
> Info : telnet server disabled
> 
> Info : CMSIS-DAP: SWD  Supported
> 
> Info : CMSIS-DAP: JTAG Supported
> 
> Info : CMSIS-DAP: FW Version = 01.27.0082
> 
> Info : CMSIS-DAP: Serial# = J41800093192
> 
> Info : CMSIS-DAP: Interface Initialised (SWD)
> 
> Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
> 
> Info : CMSIS-DAP: Interface ready
> 
> Info : clock speed 400 kHz
> 
> Info : SWD DPIDR 0x2ba01477
> 
> Info : at91samd51j19.cpu: hardware has 6 breakpoints, 4 watchpoints
> 
> Info : accepting 'gdb' connection from pipe
> 
> Error: Couldn't find part corresponding to DID 60060005
> 
> Error: auto_probe failed
> 
> Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
> 
> Error: attempted 'gdb' connection rejected
> 
> Error: error during select: Unknown error
> 
> .pioinit:12: Error in sourced command file:
> 
> Remote communication error.  Target disconnected.: No error.
> 
>

coolio986, 2019-05-30:
With the same ice connected to AS7 i can read the device signature without issue. id matches 0x60060005

ivankravets, 2019-05-30:
> Error: Couldn't find part corresponding to DID 60060005

AS7 uses own debug server. They don't use openOCD. It seems like an issue with openOCD?

coolio986, 2019-05-30:
It appears to be openOCD. I also set the debug_port to the port listed in AS7. Only doing that resulted in a program hang.

ivankravets, 2019-05-30:
Could you try to disconnect atmel ice? Do you see the same error? Maybe openOCD even does not try to communicate with debugging probe?

coolio986, 2019-05-30:
> Reading symbols from c:\Users\Anthony\Documents\PlatformIO\Projects\ILI9341_Screen\.pio\build\adafruit_metro_m4\firmware.elf...
> 
> done.
> 
> PlatformIO Unified Debugger > http://bit.ly/pio-debug
> 
> PlatformIO: Initializing remote target...
> 
> GNU MCU Eclipse OpenOCD, 64-bitOpen On-Chip Debugger 0.10.0+dev-00593-g23ad80df4 (2019-04-22-20:25)
> 
> Licensed under GNU GPL v2
> 
> For bug reports, read
> 
> 	http://openocd.org/doc/doxygen/bugs.html
> 
> at91samd51j19
> 
> Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
> 
> none separate
> 
> adapter speed: 400 kHz
> 
> cortex_m reset_config sysresetreq
> 
> Info : tcl server disabled
> 
> Info : telnet server disabled
> 
> Error: unable to find CMSIS-DAP device
> 
> 
> 
> .pioinit:12: Error in sourced command file:
> 
> Remote communication error.  Target disconnected.: No error.
> 
>

coolio986, 2019-05-30:
That tells me it is being found since the prior post it never complains about the DAP missing.

ivankravets, 2019-05-30:
Could you experiment from CLI? 
- https://www.omzlo.com/articles/programming-samd21-using-atmel-ice-with-openocd

Maybe, setting `cmsis_dap_vid_pid` will help?

coolio986, 2019-05-30:
Using windows

coolio986, 2019-05-30:
> Open On-Chip Debugger 0.10.0 (2019-04-26) [https://github.com/sysprogs/openocd]
> Licensed under GNU GPL v2
> For bug reports, read
>         http://openocd.org/doc/doxygen/bugs.html
> Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
> none separate
> adapter speed: 400 kHz
> cortex_m reset_config sysresetreq
> Info : Listening on port 6666 for tcl connections
> Info : Listening on port 4444 for telnet connections
> Error: unable to open CMSIS-DAP device 0x3eb:0x2141

coolio986, 2019-05-30:
hardware id matches device manager VID_03EB&PID_2141

coolio986, 2019-05-30:
Second time around it worked

coolio986, 2019-05-30:
> Open On-Chip Debugger 0.10.0 (2019-04-26) [https://github.com/sysprogs/openocd]
> Licensed under GNU GPL v2
> For bug reports, read
>         http://openocd.org/doc/doxygen/bugs.html
> Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
> none separate
> adapter speed: 400 kHz
> cortex_m reset_config sysresetreq
> Info : Listening on port 6666 for tcl connections
> Info : Listening on port 4444 for telnet connections
> Info : CMSIS-DAP: SWD  Supported
> Info : CMSIS-DAP: JTAG Supported
> Info : CMSIS-DAP: FW Version = 01.27.0082
> Info : CMSIS-DAP: Serial# = J41800093192
> Info : CMSIS-DAP: Interface Initialised (SWD)
> Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
> Info : CMSIS-DAP: Interface ready
> Info : clock speed 400 kHz
> Info : SWD DPIDR 0x2ba01477
> Info : at91samd51j19a.cpu: hardware has 6 breakpoints, 4 watchpoints
> Info : at91samd51j19a.cpu: external reset detected
> Info : Listening on port 3333 for gdb connections

ivankravets, 2019-05-30:
What did you change?

coolio986, 2019-05-30:
I closed AS7.

But i also tried VScode again with the same error.

coolio986, 2019-05-30:
What is the cfg path for openocd on platformio?

ivankravets, 2019-05-30:
Firstly, try to configure openOCD manually http://docs.platformio.org/en/latest/projectconf/section_env_debug.html#debug-server

If it works, I'll explain what to do next. Thanks!

coolio986, 2019-05-30:
Can't find openocd.cfg even though its in my PATH variables

coolio986, 2019-05-30:
> undefinedError: 
> Could not launch Debug Server 'C:\Users\Anthony\Downloads\OpenOCD-20190426-0.10.0\bin\openocd -f C:\Users\Anthony\Downloads\OpenOCD-20190426-0.10.0\bin\openocd.cfg'. Please check that it is installed and is included in a system PATH

coolio986, 2019-05-30:
If i try to use multi line for debug_server I end up with a parsing error

ivankravets, 2019-05-30:
Try this 
```
[env:myenv]
board = ...
debug_tool = custom
debug_port = localhost:3333
```

Please start openOCD manually in separate terminal before debugging. See other custom examples http://docs.platformio.org/en/latest/plus/debug-tools/custom.html

coolio986, 2019-05-30:
> Open On-Chip Debugger 0.10.0 (2019-04-26) [https://github.com/sysprogs/openocd]
> Licensed under GNU GPL v2
> For bug reports, read
>         http://openocd.org/doc/doxygen/bugs.html
> Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
> none separate
> adapter speed: 400 kHz
> cortex_m reset_config sysresetreq
> Info : Listening on port 6666 for tcl connections
> Info : Listening on port 4444 for telnet connections
> Info : CMSIS-DAP: SWD  Supported
> Info : CMSIS-DAP: JTAG Supported
> Info : CMSIS-DAP: FW Version = 01.27.0082
> Info : CMSIS-DAP: Serial# = J41800093192
> Info : CMSIS-DAP: Interface Initialised (SWD)
> Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
> Info : CMSIS-DAP: Interface ready
> Info : clock speed 400 kHz
> Info : SWD DPIDR 0x2ba01477
> Info : at91samd51j19a.cpu: hardware has 6 breakpoints, 4 watchpoints
> Info : Listening on port 3333 for gdb connections
> Info : accepting 'gdb' connection on tcp/3333
> Error: Couldn't find part corresponding to DID 60060005
> Error: auto_probe failed
> Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
> Error: attempted 'gdb' connection rejected

coolio986, 2019-05-30:
according to this:
[](https://github.com/arduino/OpenOCD/blob/master/src/flash/nor/at91samd.c)

There is no samd51 family (line 261):

```
/* Known SAMD families */
static const struct samd_family samd_families[] = {
	{ SAMD_PROCESSOR_M0, SAMD_FAMILY_D, SAMD_SERIES_20,
		samd20_parts, ARRAY_SIZE(samd20_parts) },
	{ SAMD_PROCESSOR_M0, SAMD_FAMILY_D, SAMD_SERIES_21,
		samd21_parts, ARRAY_SIZE(samd21_parts) },
	{ SAMD_PROCESSOR_M0, SAMD_FAMILY_D, SAMD_SERIES_21,
		samr21_parts, ARRAY_SIZE(samr21_parts) },
	{ SAMD_PROCESSOR_M0, SAMD_FAMILY_D, SAMD_SERIES_09,
		samd09_parts, ARRAY_SIZE(samd09_parts) },
	{ SAMD_PROCESSOR_M0, SAMD_FAMILY_D, SAMD_SERIES_10,
		samd10_parts, ARRAY_SIZE(samd10_parts) },
	{ SAMD_PROCESSOR_M0, SAMD_FAMILY_D, SAMD_SERIES_11,
		samd11_parts, ARRAY_SIZE(samd11_parts) },
	{ SAMD_PROCESSOR_M0, SAMD_FAMILY_L, SAMD_SERIES_21,
		saml21_parts, ARRAY_SIZE(saml21_parts) },
	{ SAMD_PROCESSOR_M0, SAMD_FAMILY_L, SAMD_SERIES_22,
		saml22_parts, ARRAY_SIZE(saml22_parts) },
	{ SAMD_PROCESSOR_M0, SAMD_FAMILY_C, SAMD_SERIES_20,
		samc20_parts, ARRAY_SIZE(samc20_parts) },
	{ SAMD_PROCESSOR_M0, SAMD_FAMILY_C, SAMD_SERIES_21,
		samc21_parts, ARRAY_SIZE(samc21_parts) },
};
```

ivankravets, 2019-05-30:
This is an old clone by arduino v0.9. Please use official source code https://repo.or.cz/w/openocd.git

ivankravets, 2019-05-30:
https://repo.or.cz/openocd.git/blob/HEAD:/src/flash/nor/atsame5.c

ivankravets, 2019-05-30:
Could you also report this issue to https://sourceforge.net/p/openocd/tickets/?

coolio986, 2019-05-30:
I am running  20190426-0.10.0 which should include this merge:
[https://github.com/gnu-mcu-eclipse/openocd/issues/37](https://github.com/gnu-mcu-eclipse/openocd/issues/37)

coolio986, 2019-05-30:
Reported to openocd:

gnu-mcu-eclipse/openocd#37

ivankravets, 2019-05-30:
@coolio986 I recommend opening a ticket at https://sourceforge.net/p/openocd/tickets/

@ilg-ul does not maintain the openOCD project and only provide binaries

ilg-ul, 2019-05-30:
If the problem is caused by some missing option in the `configure` step, please let me know and I can try a new build. I currently use:

https://github.com/gnu-mcu-eclipse/openocd-build/blob/0b960494bed41b1bf85a21b8473b7b015ab84566/scripts/container-apps-functions-source.sh#L143

If the problem is actually a functional bug, please open a ticket upstream and let me know when fixed.

coolio986, 2019-05-30:
My mistake

coolio986, 2019-05-30:
Ticket posted here:

[https://sourceforge.net/p/openocd/tickets/240/](https://sourceforge.net/p/openocd/tickets/240/)

coolio986, 2019-05-30:
@ivankravets The blob from [https://repo.or.cz/openocd.git/blob/HEAD:/src/flash/nor/atsame5.c](https://repo.or.cz/openocd.git/blob/HEAD:/src/flash/nor/atsame5.c)

doesn't match the current 0.10 from openocd:
[https://sourceforge.net/projects/openocd/files/openocd/0.10.0-rc2/](https://sourceforge.net/projects/openocd/files/openocd/0.10.0-rc2/)

coolio986, 2019-05-30:
openocd doesn't have any reference to samd51 in at91samd.c

Is there a newer version that is a compiled binary i could try?

coolio986, 2019-05-31:
openocd responded with this:

> SAM D51, E54, E53 and E51 familes were not supported by 0.10.0 release.
> 
> Use a newer OpenOCD build and target/atsame5x.cfg config.

ilg-ul, 2019-05-31:
GNU MCU Eclipse OpenOCD 0.10.0-12 uses commit c19de41a0 from Apr 7th, 2019.

If you need an even newer version, I'm afraid you have to compile it yourself.

ivankravets, 2019-05-31:
>  Apr 7th, 2019

Oh, SAM D51, E54, E53 and E51 changes were added later. @ilg-ul could you provide new binaries?

Thanks!

ivankravets, 2019-05-31:
@ilg-ul sorry, it seems that support for SAME was merged in February => https://repo.or.cz/openocd.git?a=commit;h=deaf3d264123391d8fe5c4cccbf8fb8852e1be23

Do you use official https://repo.or.cz/openocd.git for your binaries?

ilg-ul, 2019-05-31:
according to http://openocd.org/repos/, the official repo is git://git.code.sf.net/p/openocd/code.

GNU MCU Eclipse OpenOCD follows that repo.

commit deaf3d264 is from 5 February 2019 and is already included in 0.10.0-12.

are you sure you are using the correct script?

coolio986, 2019-06-05:
@ivankravets Any comment on this?

ivankravets, 2019-06-05:
In a case with openOCD for Windows we use the latest binary by @ilg-ul from https://github.com/gnu-mcu-eclipse/openocd/releases?

coolio986, 2019-06-05:
If the binary is created from the eclipse openocd commits, then samd51 should already be in there right?

ivankravets, 2019-06-07:
@coolio986 I did some update to dev/platform. Please re-test:

1. Switch to upstream version of dev/platform => http://docs.platformio.org/en/latest/platforms/atmelsam.html#upstream
2. Update all packages via `$ pio update`

Do we have any progress now?

coolio986, 2019-06-07:
Thanks @ivankravets I will test in a few hours!

coolio986, 2019-06-07:
@ivankravets I can confirm it works! Thank you for your help!

ivankravets, 2019-06-11:
Great! The final release is out => https://github.com/platformio/platform-atmelsam/releases/tag/v3.6.1

closed 2019-06-11 by ivankravets