platformio/platform-atmelsam #61: Arduino Nano 33 IOT Board Support

jamesbebbington, 2019-06-26:
I would like to please request support to be added for the recently released [Arduino Nano 33 IOT Board](https://store.arduino.cc/nano-33-iot) which is based on the [SAMD21G18A microcontroller](http://ww1.microchip.com/downloads/en/DeviceDoc/SAMD21-Family-DataSheet-DS40001882D.pdf).

Please let me know if this isn't the correct repo for this request. Many thanks.

valeros, 2019-06-27:
Hi @jamesbebbington ! This board is already added, just install this platform from the [dev branch](http://docs.platformio.org/en/latest/platforms/atmelsam.html#stable-and-upstream-versions).

closed 2019-06-27 by valeros

jamesbebbington, 2019-06-27:
Ah, I did a search of the GitHub issues but that didn't turn up any about adding support for this board. Many thanks for the info, I'll give the dev branch a go.