platformio/platform-atmelsam #62: Compiling a C++ sketch for Arduino DUE gives syntax errors

LorenzoDeMarco, 2019-06-27:
I tried to compile a previously working Arduino DUE sketch with the latest 'atmelsam' updates from PlatformIO (latest version).. but syntax errors pop out when compiling, all regarding `#define Max(a,b) (((a) > (b)) ? (a) : (b))` from the library source files: `Expected unqualified-id before '(' token`
The macro itself is correct, and I also found out that removing any occurrence of `std::map` in my code removes the errors... is there a solution for this issue?
Thanks in advance
Lorenzo
[image]

valeros, 2019-06-27:
Does it work with the previous version of the platform?

LorenzoDeMarco, 2019-06-27:
No, It doesn't. Attempting to use std::map causes the above messages to appear.
I tried to wrap all std::map members in conditional precompiler blocks as a test, and it seems that removing the std::maps from the code this way allows me to compile the sketch with no errors...

valeros, 2019-06-28:
Hmm, I couldn't reproduce the issue. Could you put together a minimal example?

closed 2019-07-11 by ivankravets