platformio/platform-atmelsam #64: [Question] samd51 and system init

GMagician, 2019-07-09:
In samd51/Grand Central M4 startup is missing ADC and USB calib data load (copy some values from calibration flash area to registers).

I would like to add it but:
Startup code seems to comes from Adafruit but in adafruit repo is missing your "system" (that comes from atmel I think).
"system" folder defines all the registers where to store calibration data, but misses the ones where to get such values (0x800080).

The best is to add some define in atmel repo and some in adafruit repo....or add them here that is the collecting place.

What is the best approach?

ivankravets, 2019-07-11:
Could you reproduce this issue with the latest release?

GMagician, 2019-07-11:
I have uploaded fix to adafruit repository, but it takes a a long time to be merged

ivankravets, 2019-07-11:
Could you share an issue URL here?

GMagician, 2019-07-11:
here it is https://github.com/adafruit/ArduinoCore-samd/pull/127

ivankravets, 2019-07-11:
Thanks. Please ping us here when @ladyada will merge it and release a new version of their core.

GMagician, 2019-07-11:
ok

GMagician, 2019-08-22:
it seems it has been added today.
From usb side it was not needed because init was already done in usb class (but I think that it's better done in init stage).
From adc side, even if no difference has been noted, atmel say to do it, so...

GMagician, 2019-09-01:
[ot] @ivankravets one question. platform atmelsam released from pio (used by adafruit grand central m4) uses this repo? this repo is aligned with adafruit one? which branch? I ask because I see some fix I made there (variant.h) that are missing in released pio one

GMagician, 2020-01-05:
@ivankravets adafruid has updated its framework to version 1.5.9.
This new release include 'init' sequence and also a fix in tinyusb that prevent Marlin firmware to properly works with grandcentral_m4 (samd51)

closed 2020-01-05 by GMagician