platformio/platform-atmelsam #66: [BUG] Samd51 upload code

GMagician, 2019-07-11:
I'm facing an error while uploading code to samd51. 

As described in https://community.platformio.org/t/samd51-upload-code-error/8317,
using platformio, flash is correctly erased by I get an error while writing code

here is error I get:

> Erase flash
chipErase(addr=0x4000)
Done in 5.001 seconds
Write 62264 bytes to flash (122 pages)
[ ] 0% (0/122 pages)w
rite(addr=0SAM-BA operation failed
x20004034,size=0x1000)
writeBuffer(scr_addr=0x20004034, dst_addr=0x4000, size=0x1000)
scons: *** [upload] Error 1
[ERROR] Took 42.20 seconds

ivankravets, 2019-07-11:
Could you enable verbose mode in Arduino IDE settings and provide an upload output from Arduino IDE? Need to see which upload flags do they use.

GMagician, 2019-07-11:
This is what I get with verbose enabled

ivankravets, 2019-07-11:
Then compare with `platformio run -t upload -v`.

GMagician, 2019-07-11:
This is PIO cmd line

> bossac --info --debug --port "COM8" --erase --write --verify --reset -U --offset 0x4000 .pioenvs\adafruit_grandcentral_m4\firmware.bin

and this is arduino one:

> bossac -i -d --port=COM8 -U -i --offset=0x4000 -w -v C:\Users\Giuliano\AppData\Local\Temp\arduino_build_683858/Marlin.ino.bin -R

GMagician, 2019-07-11:
--erase is the problem

ivankravets, 2019-07-11:
Thanks for the report! Does it work with upstream version http://docs.platformio.org/en/latest/platforms/atmelsam.html#upstream?

closed 2019-07-11 by ivankravets

GMagician, 2019-07-11:
I replace file in my folder and it works.
I suspect there is an issue in bootloader...I'll dig into that code

commit cbf174fd (2019-07-11): Do not erase flash before uploading for Adafruit boards // Resolve #66