platformio/platform-atmelsam #67: USE_TINYUSB build flag shouldn't be set by default for all boards

SRGDamia1, 2019-07-12:
On the Arduino IDE it is used if selected by menu, not by default: https://github.com/adafruit/Adafruit_TinyUSB_Arduino/issues/13

erikkallen, 2019-07-23:
I could not compile usb host library samd anymore and had to add 
    build_flags = -UUSE_TINYUSB
to platformio.ini as a workaround hope it helps anyone

closed 2019-09-02 by valeros

deladriere, 2019-10-01:
@valeros  Ok now on atmelsam@3.8.1 the tinyusb is not set by default.
What is I need to use the tinyusb : what would be the build_flags syntax ? 
can I use build_flags = -D USE_TINYUSB

valeros, 2019-10-01:
`build_flags = -D USE_TINYUSB` - that's correct.

deladriere, 2019-10-01:
Ok the flags seem to be  working  but compiling this example [https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/examples/MIDI/midi_test/midi_test.ino](url)
gives this error 

`src/main.cpp: In function 'void setup()':
src/main.cpp:55:24: error: 'handleNoteOn' was not declared in this scope
   MIDI.setHandleNoteOn(handleNoteOn);
                        ^~~~~~~~~~~~
src/main.cpp:58:25: error: 'handleNoteOff' was not declared in this scope
   MIDI.setHandleNoteOff(handleNoteOff);`

`[env:adafruit_feather_m0]
platform = atmelsam@3.8.1
board = adafruit_feather_m0
build_flags = -D USE_TINYUSB
framework = arduino
lib_deps =
    Adafruit TinyUSB Library
    MIDI Library`

it compiles and runs fine on the Arduino IDE

valeros, 2019-10-01:
That problem is with the code, if you are using `.cpp` extension then you need to convert the example to valid `C++` code [manually](https://docs.platformio.org/en/latest/faq.html#convert-arduino-file-to-c-manually). So `handleNoteOff` function should be at least declared above the first use.

deladriere, 2019-10-01:
Ah sorry! I corrected the code by declaring the function before use
but I still have an error : `src/main.cpp:25:20: error: cannot declare variable 'usb_midi' to be of abstract type 'Adafruit_USBD_MIDI'`

valeros, 2019-10-01:
That's more subtle issue, what version of Atmel SAM package are you using in Arduino IDE? What version of `Adafruit TinyUSB` is installed there?

deladriere, 2019-10-01:
Adadruit Samd 1.5.3  with [Adafruit_TinyUSB_Library@0.6.0]

valeros, 2019-10-02:
The outdated adafruit core was the problem. Please update to the latest v3.9.0 and try again. Thanks!

deladriere, 2019-10-03:
Thanks but this gives a new error
`Processing adafruit_feather_m0 (platform: atmelsam@3.9.0; board: adafruit_feather_m0; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------
PlatformManager: Installing atmelsam @ 3.9.0
Error: Could not find a version that satisfies the requirement '3.9.0' for your system 'darwin_x86_64'
The terminal process terminated with exit code: 1`

valeros, 2019-10-03:
@deladriere Yeah, sorry, 3.9.0 is released now, please try again.

deladriere, 2019-10-03:
Its working fine now, thanks!

commit d0ce0769 (2019-09-02): Don't use TinyUSB stack by default with Adafruit core // Resolve #67, Resolve #69