platformio/platform-atmelsam #68: version 3.7.0 : Warning in 3.6.1 becomes error in 3.7.0

deladriere, 2019-07-14:
The same line creates an error in 3.7.0 (and also in 3.8.0)
I can't compile anymore. What can I do?
thanks 

@3.6.1

int8_t tmsK10[0x08] = {0xCD, 0xDF, 0xF1, 0x04, 0x16, 0x20, 0x3B, 0x4D};
^
src/talkie.cpp:52:70: warning: narrowing conversion of '223' from 'int' to 'int8_t {aka signed char}' inside { } [-Wnarrowing]
src/talkie.cpp:52:70: warning: narrowing conversion of '241' from 'int' to 'int8_t {aka signed char}' inside { } [-Wnarrowing]

@3.7.0

int8_t tmsK10[0x08] = {0xCD, 0xDF, 0xF1, 0x04, 0x16, 0x20, 0x3B, 0x4D};
^
src/talkie.cpp:52:70: error: narrowing conversion of '223' from 'int' to 'int8_t {aka signed char}' inside { } [-Wnarrowing]
src/talkie.cpp:52:70: error: narrowing conversion of '241' from 'int' to 'int8_t {aka signed char}' inside { } [-Wnarrowing]

pbolduc, 2019-07-16:
If you want to ignore the warning/error, you could try adding ```-Wno-narrowing``` to your build_flags.

deladriere, 2019-10-13:
I have the same problem with @3.9.0
I can work it around with `build_flags = -Wno-narrowing`
Is it safe to work like that?

valeros, 2020-02-12:
Hi @deladriere ! Looks like it's only related to your project, so `build_flags = -Wno-narrowing` is fine.

closed 2020-02-12 by valeros