platformio/platform-atmelsam #69: atmelsam@3.8.0  won't compile MIDIUSB@1.0.4

deladriere, 2019-07-14:
It works fine with atmelsam@3.7.0
here is my Platformio.ini

[env:adafruit_feather_m0]
platform = atmelsam@3.8.0
board = adafruit_feather_m0
framework = arduino

lib_deps =
  # Using a library name
  MIDIUSB@1.0.4
  SSD1306Ascii@1.2.4

It gives many errors 
.platformio/packages/framework-arduinosam/cores/adafruit/USB/USBAPI.h:118:23: error: conflicting declaration 'USBDeviceClass USBDevice'
.platformio/packages/framework-arduinosam/cores/adafruit/USB/USBAPI.h:189:16: error: conflicting declaration 'Serial_ Serial'
.platformio/packages/framework-arduinosam/cores/adafruit/USB/USBAPI.h:118:23: error: conflicting declaration 'USBDeviceClass USBDevice'

smp4488, 2019-07-27:
@deladriere I ran into this issue with my SparkFun SamD21 Mini. I had to go into the library files mentioned in the errors and update references to `Serial` to be `SerialUSB`. This is because the default serial on most Arduinos is `Serial` and on the SamD21 it is `SerialUSB` for usb serial. I haven't been able to find a good work around besides editing the library manually.

valeros, 2019-09-02:
Hi @deladriere ! Thanks for reporting the issue. Please try the latest platform from the development branch:
```
[env:adafruit_feather_m0]
platform = https://github.com/platformio/platform-atmelsam.git
board = adafruit_feather_m0
framework = arduino

lib_deps =
  MIDIUSB@1.0.4
  SSD1306Ascii@1.2.4
```

closed 2019-09-02 by valeros

deladriere, 2019-09-04:
hi @valeros 
this version
`platform = https://github.com/platformio/platform-atmelsam.git`
 is working fine !
thanks a lot

commit d0ce0769 (2019-09-02): Don't use TinyUSB stack by default with Adafruit core // Resolve #67, Resolve #69