platformio/platform-atmelsam #78: Arduino Due does not excute code after upload

aurror, 2019-09-02:
Hey,

it seems like there is a bug related to the arduino due in the current version.
The upload works, but the uploaded code does not execute and I can't open the serial console. 
If I upload the same code from arduino IDE it works. 
It also works with an older option of atmelsam: 
`platform = atmelsam@3.1.2`

I'm using platformio version 4.0.3

valeros, 2019-09-03:
Were there any issues with uploading? Could you please provide upload log?

aurror, 2019-09-08:
There were no issues while uploading. I noticed this error when uploading from a raspberry pi 3B with a clean installation.
Unfortunately I can't provide the log since it works with the old version and I currently have no time to do testing.

ivankravets, 2019-09-16:
Do you use USB or Programming port?

aurror, 2019-09-19:
I used the programming port.

aurror, 2019-12-11:
Just tested it again on a new raspberry pi 3b / arduino due, still got the same issue.
Used version: Atmel SAM 3.9.0

Log:
```
Processing due (platform: atmelsam; board: sainSmartDue; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelsam/sainSmartDue.html
PLATFORM: Atmel SAM 3.9.0 > SainSmart Due (Programming Port)
HARDWARE: AT91SAM3X8E 84MHz, 96KB RAM, 512KB Flash
DEBUG: Current (atmel-ice) External (atmel-ice, blackmagic, jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinosam 4.4.191002, tool-bossac 1.10700.190624 (1.7.0)
Converting Repetier.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 3 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SPI> 1.0
Building in release mode
Compiling .pio/build/due/src/BedLeveling.cpp.o
Compiling .pio/build/due/src/Commands.cpp.o
Compiling .pio/build/due/src/Communication.cpp.o
Compiling .pio/build/due/src/Distortion.cpp.o
Compiling .pio/build/due/src/Drivers.cpp.o
Compiling .pio/build/due/src/Eeprom.cpp.o
Compiling .pio/build/due/src/Endstops.cpp.o
Compiling .pio/build/due/src/Extruder.cpp.o
Compiling .pio/build/due/src/HAL.cpp.o
src/Commands.cpp: In static member function 'static void Commands::executeGCode(GCode*)':
src/Commands.cpp:2922:29: warning: 'actSource' may be used uninitialized in this function [-Wmaybe-uninitialized]
   GCodeSource::activeSource = actSource;
   ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
Compiling .pio/build/due/src/Printer.cpp.o
Compiling .pio/build/due/src/Repetier.ino.cpp.o
src/HAL.cpp: In static member function 'static void HAL::i2cSetClockspeed(uint32_t)':
src/HAL.cpp:543:58: warning: 'dwClDiv' may be used uninitialized in this function [-Wmaybe-uninitialized]
     TWI_INTERFACE->TWI_CWGR = (dwCkDiv << 16) | (dwClDiv << 8) | dwClDiv;
                                                 ~~~~~~~~~^~~~~
Compiling .pio/build/due/src/SDCard.cpp.o
Compiling .pio/build/due/src/SdFat.cpp.o
Compiling .pio/build/due/src/gcode.cpp.o
Compiling .pio/build/due/src/motion.cpp.o
Compiling .pio/build/due/src/ui.cpp.o
Compiling .pio/build/due/src/uilang.cpp.o
Compiling .pio/build/due/libe6e/SPI/SPI.cpp.o
Compiling .pio/build/due/FrameworkArduinoVariant/variant.cpp.o
Archiving .pio/build/due/libe6e/libSPI.a
Indexing .pio/build/due/libe6e/libSPI.a
Compiling .pio/build/due/FrameworkArduino/IPAddress.cpp.o
Compiling .pio/build/due/FrameworkArduino/Print.cpp.o
Archiving .pio/build/due/libFrameworkArduinoVariant.a
Indexing .pio/build/due/libFrameworkArduinoVariant.a
Compiling .pio/build/due/FrameworkArduino/Reset.cpp.o
Compiling .pio/build/due/FrameworkArduino/RingBuffer.cpp.o
Compiling .pio/build/due/FrameworkArduino/Stream.cpp.o
Compiling .pio/build/due/FrameworkArduino/UARTClass.cpp.o
Compiling .pio/build/due/FrameworkArduino/USARTClass.cpp.o
Compiling .pio/build/due/FrameworkArduino/USB/CDC.cpp.o
Compiling .pio/build/due/FrameworkArduino/USB/PluggableUSB.cpp.o
Compiling .pio/build/due/FrameworkArduino/USB/USBCore.cpp.o
Compiling .pio/build/due/FrameworkArduino/WInterrupts.c.o
Compiling .pio/build/due/FrameworkArduino/WMath.cpp.o
Compiling .pio/build/due/FrameworkArduino/WString.cpp.o
Compiling .pio/build/due/FrameworkArduino/abi.cpp.o
/home/pi/.platformio/packages/framework-arduinosam/cores/sam/USB/USBCore.cpp: In function 'void USB_ISR()':
/home/pi/.platformio/packages/framework-arduinosam/cores/sam/USB/USBCore.cpp:688:21: warning: this 'else' clause does not guard... [-Wmisleading-indentation]
                     else
                     ^~~~
[... 143 lines cut ...]
DATA:    [====      ]  41.1% (used 13472 bytes from 32768 bytes)
PROGRAM: [==        ]  22.7% (used 119064 bytes from 524288 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, jlink, sam-ba
CURRENT: upload_protocol = sam-ba
Looking for upload port...
Use manually specified: /dev/ttyACM0
Forcing reset using 1200bps open/close on port /dev/ttyACM0
Uploading .pio/build/due/firmware.bin
Atmel SMART device 0x285e0a60 found
Erase flash
done in 0.040 seconds

Write 122024 bytes to flash (477 pages)
[==============================] 100% (477/477 pages)
done in 29.296 seconds

Verify 122024 bytes of flash
[==============================] 100% (477/477 pages)
Verify successful
done in 25.236 seconds
Set boot flash true
CPU reset.
======================== [SUCCESS] Took 175.74 seconds ========================
```

tfeldmann, 2020-03-24:
We had the same problem and solved it by switching back to bossac 1.10700.190624. The newer version 1.10900.0 did not work for us.

Links:
https://dl.bintray.com/platformio/dl-packages/tool-bossac-linux_armv6l-1.10700.190624.tar.gz
https://dl.bintray.com/platformio/dl-packages/tool-bossac-linux_armv6l-1.10900.0.tar.gz

valeros, 2020-03-24:
Hi @aurror ! Can you confirm that it works with `1.10601.0` for you?  Just specify required version in your `platformio.ini`:
```
platform_packages =
  ; use bossac 1.6.1
  tool-bossac@~1.10601.0
```

RudolphRiedel, 2020-11-29:
I recently started playing with an Adafruit Metro M4 and quickly noticed that I can only program it once with PlatformIO.
The second attempt to flash the board fails and the only way to get it back working was to flash the board with the ArduinoIDE, then I could flash it again once with PlatformIO.
This is due to the included version of BOSSA.
I manually replaced the bossac.exe in \.platformio\packages\tool-bossac with version 1.9.1 from
here: https://github.com/shumatech/BOSSA
And it worked just fine untill the update from 5.0.0 to 5.0.1 of Atmel SAM "repaired" it.
I just replaced the file again and manually and it is working fine again now.