platformio/platform-atmelsam #80: CMSIS build

frankleonrose, 2019-10-18:
In addition to #33, I'd like to build for my SAMD21g boards using `framework = cmsis`. Where can I find the repo for that?

frankleonrose, 2019-10-19:
I created https://github.com/frankleonrose/framework-cmsis from scratch and started my work there. @valeros, where is the actual repo I should fork from?

valeros, 2019-10-22:
Hi @frankleonrose ! Sorry for the late reply. 
> Where is the actual repo I should fork from?

At the moment there is no repository for this package, so your https://github.com/frankleonrose/framework-cmsis is totally fine you just need to specify the url to this repository it in version field in `platform.json` file. Later when you finish the build script, we'll add it to our packages.

RudolphRiedel, 2020-04-14:
As this is exactly what I was looking for, what is the reason that the Pull-Request has not been applied in early november after the cleanup of the STM32 leftovers had been done?

maxgerhardt, 2021-08-06:
This was asked again in https://community.platformio.org/t/bare-metal-on-arduino-due-atmelsam/22824