platformio/platform-atmelsam #86: Arduino MKR WAN 1310

03sarath, 2019-11-16:
Requesting to add Arduino MKR WAN 1310 board for Atmel SAM section.

remuslazar, 2020-01-12:
Because somebody asked me how to leverage https://github.com/platformio/platform-atmelsam/pull/90, here a quick "how to":

You have to use my fork instead of the official repository. In you `platformio.ini` file, replace this line

```
platform = atmelsam
```

with this one:

```
platform = https://github.com/remuslazar/platform-atmelsam
```

(I already merged the needed changes into my default branch, so everything should be fine).

Then, do a `platformio update` and `platformio run`. Then `platformio boards` should also find the mkrwan1310.

And: don't forget to revert it when #90 is merged!

valeros, 2020-02-12:
Resolved in afd391496d7125237768f6f00f86fbc5098079ca. Please re-test with upstream version http://docs.platformio.org/en/latest/platforms/atmelsam.html#upstream

closed 2020-02-12 by valeros