platformio/platform-atmelsam #87: Flood of kernel log messages (xhci_hcd) after upload finishes

BitFlipp, 2019-11-24:
I'm not sure whether this is an issue with PlatformIO or with my setup. Uploading to an Arduino Zero compatible board seems to work flawlessly with PlatformIO 4.0.2 on 5.3.12-arch1-1:

```
Processing zeroUSB (platform: atmelsam; board: zeroUSB; framework: arduino)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelsam/zeroUSB.html
PLATFORM: Atmel SAM 3.9.0 > Arduino Zero (USB Native Port)
HARDWARE: SAMD21G18A 48MHz, 32KB RAM, 256KB Flash
DEBUG: Current (atmel-ice) External (atmel-ice, blackmagic, jlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinosam 4.4.191002, tool-bossac 1.10700.190624 (1.7.0)
Converting Unit.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 18 compatible libraries
Scanning dependencies...
Dependency Graph
[...omitted...]
Compiling .pio/build/zeroUSB/src/Unit.ino.cpp.o
Linking .pio/build/zeroUSB/firmware.elf
Checking size .pio/build/zeroUSB/firmware.elf
Building .pio/build/zeroUSB/firmware.bin
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [==        ]  15.4% (used 5060 bytes from 32768 bytes)
PROGRAM: [===       ]  34.2% (used 89600 bytes from 262144 bytes)
Configuring upload protocol...
AVAILABLE: atmel-ice, blackmagic, jlink, sam-ba
CURRENT: upload_protocol = sam-ba
Looking for upload port...

Warning! Your `/etc/udev/rules.d/99-platformio-udev.rules` are outdated. Please update or reinstall them.
 Mode details: https://docs.platformio.org/en/latest/faq.html#platformio-udev-rules

Auto-detected: /dev/ttyACM0
Forcing reset using 1200bps open/close on port /dev/ttyACM0
Waiting for the new upload port...
Uploading .pio/build/zeroUSB/firmware.bin
Atmel SMART device 0x10010005 found
Erase flash
done in 0.843 seconds

Write 89600 bytes to flash (1400 pages)

[=                             ] 4% (64/1400 pages)
[==                            ] 9% (128/1400 pages)
[====                          ] 13% (192/1400 pages)
[=====                         ] 18% (256/1400 pages)
[======                        ] 22% (320/1400 pages)
[========                      ] 27% (384/1400 pages)
[=========                     ] 32% (448/1400 pages)
[==========                    ] 36% (512/1400 pages)
[============                  ] 41% (576/1400 pages)
[=============                 ] 45% (640/1400 pages)
[===============               ] 50% (704/1400 pages)
[================              ] 54% (768/1400 pages)
[=================             ] 59% (832/1400 pages)
[===================           ] 64% (896/1400 pages)
[====================          ] 68% (960/1400 pages)
[=====================         ] 73% (1024/1400 pages)
[=======================       ] 77% (1088/1400 pages)
[========================      ] 82% (1152/1400 pages)
[==========================    ] 86% (1216/1400 pages)
[===========================   ] 91% (1280/1400 pages)
[============================  ] 96% (1344/1400 pages)
[==============================] 100% (1400/1400 pages)
done in 0.481 seconds

Verify 89600 bytes of flash with checksum.
Verify successful
done in 0.079 seconds
CPU reset.
```

After the upload finishes, my kernel log is flooded with messages like these for several minutes:

```
[ 1172.488387] xhci_hcd 0000:00:14.0: A Set TR Deq Ptr command is pending.
[ 1172.492722] xhci_hcd 0000:00:14.0: WARN Cannot submit Set TR Deq Ptr
[ 1172.492725] xhci_hcd 0000:00:14.0: A Set TR Deq Ptr command is pending.
[ 1172.505232] xhci_hcd 0000:00:14.0: WARN Cannot submit Set TR Deq Ptr
[ 1172.505236] xhci_hcd 0000:00:14.0: A Set TR Deq Ptr command is pending.
[ 1172.524984] xhci_hcd 0000:00:14.0: WARN Cannot submit Set TR Deq Ptr
[ 1172.524988] xhci_hcd 0000:00:14.0: A Set TR Deq Ptr command is pending.
```

I would like to avoid them, but I really have no idea what they mean or what causes them.

BitFlipp, 2019-11-24:
I could reproduce this with PlatformIO 4.1.0 and a minimalistic "blink" sample.

valeros, 2020-02-12:
Hi @BitFlipp ! Could you please try to upload a simple blink example using Arduino IDE?