platformio/platform-atmelsam #89: pls update to include adafruit ArduinoCore-samd 1.5.9

deladriere, 2020-01-08:


valeros, 2020-02-12:
Resolved in abbfb1b6ca74c27e8ca6e47e9132a4c31c883d05. Please re-test with upstream version http://docs.platformio.org/en/latest/platforms/atmelsam.html#upstream

closed 2020-02-12 by valeros

deladriere, 2020-02-12:
@valeros I still have a compile error
Here is my platformio.ini

[env:upstream_develop]
platform = https://github.com/platformio/platform-atmelsam.git
board = adafruit_feather_m0
build_flags = -D USE_TINYUSB
framework = arduino
lib_deps =
    Adafruit TinyUSB Library@0.6.0
    MIDI Library@4.3.1
    SSD1306Ascii@1.2.4
monitor_speed = 115200

I Still have this error :
src/main.cpp: In function 'void setup()':
src/main.cpp:376:12: error: 'class Adafruit_USBD_Device' has no member named 'setManufacturerDescriptor'
  USBDevice.setManufacturerDescriptor("Polaxis");
            ^~~~~~~~~~~~~~~~~~~~~~~~~
src/main.cpp:377:12: error: 'class Adafruit_USBD_Device' has no member named 'setProductDescriptor'
  USBDevice.setProductDescriptor("Robovox");
            ^~~~~~~~~~~~~~~~~~~~

It works fine with the Arduino IDE

valeros, 2020-02-12:
Hi @deladriere ! Just tried a basic code snippet with `USBDevice.setManufacturerDescriptor("Polaxis");` and everything works just fine. Could you please compile your project in verbose mode and attach here the entire log?

deladriere, 2020-02-12:
hi @valeros 
here is the output

Robovox_Emy_midi_tiny *****$ pio run -v
Processing upstream_develop (platform: https://github.com/platformio/platform-atmelsam.git; board: adafruit_feather_m0; build_flags: -D USE_TINYUSB; framework: arduino; lib_deps: Adafruit TinyUSB Library@0.6.0, MIDI Library@4.3.1, SSD1306Ascii@1.2.4; monitor_speed: 115200)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/atmelsam/adafruit_feather_m0.html
PLATFORM: Atmel SAM 3.9.0 #204ace4 (git+https://github.com/platformio/platform-atmelsam.git) > Adafruit Feather M0
HARDWARE: SAMD21G18A 48MHz, 32KB RAM, 256KB Flash
DEBUG: Current (atmel-ice) External (atmel-ice, blackmagic, jlink)
PACKAGES: 
 - framework-arduinosam 4.4.191002 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Adafruit TinyUSB Library> 0.6.0 (/Volumes/T5/Dropbox/Elektro/git/Euro_Modules/Robo_vox/Software/Robovox_Emy_midi_tiny/.pio/libdeps/upstream_develop/Adafruit TinyUSB Library_ID6465)
|-- <MIDI Library> 4.3.1 (/Volumes/T5/Dropbox/Elektro/git/Euro_Modules/Robo_vox/Software/Robovox_Emy_midi_tiny/.pio/libdeps/upstream_develop/MIDI Library_ID62)
|-- <SSD1306Ascii> 1.2.4 (/Volumes/T5/Dropbox/Elektro/git/Euro_Modules/Robo_vox/Software/Robovox_Emy_midi_tiny/.pio/libdeps/upstream_develop/SSD1306Ascii_ID5169)
|   |-- <Wire> 1.0 (/Users/*****/.platformio/packages/framework-arduinosam/libraries/__cores__/adafruit/Wire)
|-- <MCP23008> (/Volumes/T5/Dropbox/Elektro/git/Euro_Modules/Robo_vox/Software/Robovox_Emy_midi_tiny/lib/MCP23008)
|   |-- <Wire> 1.0 (/Users/*****/.platformio/packages/framework-arduinosam/libraries/__cores__/adafruit/Wire)
|-- <SDU> 1.0.0 (/Users/*****/.platformio/packages/framework-arduinosam/libraries/__cores__/adafruit/SDU)
|-- <Wire> 1.0 (/Users/*****/.platformio/packages/framework-arduinosam/libraries/__cores__/adafruit/Wire)
Building in release mode
arm-none-eabi-g++ -o .pio/build/upstream_develop/src/main.cpp.o -c -fno-rtti -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -Os -ffunction-sections -fdata-sections -Wall -mthumb -nostdlib --param max-inline-insns-single=500 -mcpu=cortex-m0plus -DPLATFORMIO=40200 -DARDUINO_ARCH_SAMD -DARDUINO_SAMD_ZERO -DARDUINO_SAMD_FEATHER_M0 -DARM_MATH_CM0PLUS -D__SAMD21G18A__ -DUSE_TINYUSB -DF_CPU=48000000L -DUSBCON -DARDUINO=10805 -DUSB_VID=0x239A -DUSB_PID=0x800B "-DUSB_PRODUCT=\"Adafruit Feather M0\"" -DUSB_MANUFACTURER=\"Adafruit\" -Iinclude -Isrc -I/Users/*****/.platformio/packages/framework-arduinosam/libraries/__cores__/adafruit/SDU/src -Ilib/MCP23008 -I.pio/libdeps/upstream_develop/SSD1306Ascii_ID5169/src -I/Users/*****/.platformio/packages/framework-arduinosam/libraries/__cores__/adafruit/Wire "-I.pio/libdeps/upstream_develop/MIDI Library_ID62/src" "-I.pio/libdeps/upstream_develop/Adafruit TinyUSB Library_ID6465/src" -I/Users/*****/.platformio/packages/framework-arduinosam/cores/adafruit -I/Users/*****/.platformio/packages/framework-arduinosam/system/samd/CMSIS/CMSIS/Include -I/Users/*****/.platformio/packages/framework-arduinosam/system/samd/CMSIS-Atmel/CMSIS/Device/ATMEL -I/Users/*****/.platformio/packages/framework-arduinosam/cores/adafruit/Adafruit_TinyUSB_Core -I/Users/*****/.platformio/packages/framework-arduinosam/cores/adafruit/Adafruit_TinyUSB_Core/tinyusb/src -I/Users/*****/.platformio/packages/framework-arduinosam/variants/feather_m0 src/main.cpp
src/main.cpp: In function 'void setup()':
src/main.cpp:376:12: error: 'class Adafruit_USBD_Device' has no member named 'setManufacturerDescriptor'
  USBDevice.setManufacturerDescriptor("Polaxis");
            ^~~~~~~~~~~~~~~~~~~~~~~~~
src/main.cpp:377:12: error: 'class Adafruit_USBD_Device' has no member named 'setProductDescriptor'
  USBDevice.setProductDescriptor("Robovox");
            ^~~~~~~~~~~~~~~~~~~~
*** [.pio/build/upstream_develop/src/main.cpp.o] Error 1
======================================================================== [FAILED] Took 3.02 seconds ========================================================================

valeros, 2020-02-12:
You're using an outdated upstream version of the platform, try to run `pio update` in the terminal window or delete `/Users/your_user/.platformio/platforms/atmelsam@any_hash` directory.

deladriere, 2020-02-12:
I thought 
[env:upstream_develop]
platform = https://github.com/platformio/platform-atmelsam.git
pio update gave me some errors
I deleted the all the /Users/your_user/.platformio/platforms/atmelsam@any_hash directories
Now it compile just fine ! 
Many thanks