platformio/platform-atmelsam #91: The code is not correctly offset in flash for a Zephyr setup (at least on Adafruit Feather M0)

KubaFYI, 2020-01-15:
I was trying to get the Zephyr [blinky example](https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/basic/blinky/src/main.c) work on an Adafruit Feather M0 and noticed it doesn't work straight out of the box.

I've been able to narrow the source of the problem down to the linker scripts generating a linker command file which doesn't offsetting the program memory by correctly (it's assumed to start at 0x0 while the Feather's bootloader needs it to start at 0x2000). Changing line 4 of linker_pass_final.cmd from `FLASH (rx) : ORIGIN = (0x0 + 0x0), LENGTH = (256*1K - 0x0)` to `FLASH (rx) : ORIGIN = (0x2000 + 0x0), LENGTH = (256*1K - 0x2000)` and building fixes it. This should probably be a default configuration.

I've tried making it so but it seems I'm not familiar enough with Zephyr or the way it integrates into PlatformIO to figure out where things go wrong. The `adafruit_feather_m0_basic_proto` device tree correctly describes the flash code partition as living at the address 2000. Maybe that's not sufficient?

KubaFYI, 2020-01-21:
This seems to be related to a more general problem with zephyr not flashing Adafruit Feather M0's out of the box. See zephyrproject-rtos/zephyr#22062.

liquiddandruff, 2021-03-14:
This is resolved now right? Will be getting a Feather M0 RFM95 soon and would like it very much if I can flash :).

I'll report back if https://github.com/zephyrproject-rtos/zephyr/pull/22180 fixes this issue and if so then this can be closed.