platformio/platform-atmelsam #93: Update Sodaq Arduino Core

maxgerhardt, 2020-01-23:
Currently the Sodaq core was last updated in october 2018 and is on version 1.6.18

https://github.com/platformio/platformio-pkg-framework-arduinosam/blob/6537a952415f1f1e4e40485b190e26620e7faa81/platform_sodaq.txt#L23

The current version given in http://downloads.sodaq.net/package_sodaq_samd_index.json is 1.6.20, downloadable at http://downloads.sodaq.net/sodaqsamdboards-1.6.20.tar.bz2. PlatformIO should update to the newest version as requested in https://community.platformio.org/t/how-will-the-sodaq-core-be-updated/11633 :)

keestux, 2020-01-26:
@ivankravets why was the issue moved to platform-atmelsam? It seems to me that changes are required in platformio-pkg-framework-arduinosam.

ivankravets, 2020-01-28:
We will drop `platformio-pkg-framework-arduinosam` repository. We did the same with AVR. Now, every Arduino core has own package. @valeros will help us here soon.

keestux, 2020-01-28:
If you let us know how you want to do that, then perhaps we can help

valeros, 2020-02-12:
Resolved in abbfb1b6ca74c27e8ca6e47e9132a4c31c883d05. Please re-test with upstream version http://docs.platformio.org/en/latest/platforms/atmelsam.html#upstream

closed 2020-02-12 by valeros