platformio/platform-atmelsam #94: Error: invalid subcommand "write_image erase "

CHTUZKI, 2020-02-05:
`Processing dueUSB (platform: atmelsam; board: dueUSB; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelsam/dueUSB.html
PLATFORM: Atmel SAM 4.0.0 > Arduino Due (USB Native Port)
HARDWARE: AT91SAM3X8E 84MHz, 96KB RAM, 512KB Flash
DEBUG: Current (stlink) External (atmel-ice, blackmagic, jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinosam 4.4.191002, tool-openocd 2.1000.190707 (10.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 3 compatible libraries
Scanning dependencies...    
No dependencies
Building in release mode
Checking size .pio\build\dueUSB\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [          ]   2.6% (used 2576 bytes from 98304 bytes)  
PROGRAM: [          ]   2.0% (used 10536 bytes from 524288 bytes)
Configuring upload protocol...
AVAILABLE: atmel-ice, blackmagic, jlink, sam-ba, stlink
CURRENT: upload_protocol = stlink
Uploading .pio\build\dueUSB\firmware.bin
xPack OpenOCD, 64-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-11:28)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

0x2ba01477
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00080be8 msp: 0x20088000
** Programming Started **
flash
  flash bank bank_id driver_name base_address size_bytes chip_width_bytes
            bus_width_bytes target [driver_options ...]
  flash banks
  flash init
  flash list
gdb_flash_program ('enable'|'disable')
mflash
  mflash bank soc_type base_addr pin_id target
  mflash init
nand
program <filename> [address] [verify] [reset] [exit]
Error: invalid subcommand "write_image erase .pio\build\dueUSB\firmware.bin"
embedded:startup.tcl:449: Error: ** Programming Failed **
in procedure 'program' 
in procedure 'program_error' called at file "embedded:startup.tcl", line 508
at file "embedded:startup.tcl", line 449
*** [upload] Error 1
======================================================================================== [FAILED] Took 2.69 seconds ========================================================================================`

valeros, 2020-02-12:
Hi @CHTUZKI !  Looks like these commands `write_image erase` are specific for the project? Could you please provide more details?

fdinardo-unsam, 2021-09-18:
I have the same error... Could you solve this error? Thanks