platformio/platform-atmelsam #102: Add support for Seeeduino XIAO board

dawidowski, 2020-04-01:
I would like to ask to add support for **Seeeduino XIAO** board.

MCU: **SAMD21G18-MU** (Atmel)
MCU Datasheet PDF: https://github.com/SeeedDocument/Seeeduino-XIAO/raw/master/res/ATSAMD21G18A-MU-Datasheet.pdf

Link to board specs: http://wiki.seeedstudio.com/Seeeduino-XIAO/

Board connections schematics: https://github.com/SeeedDocument/Seeeduino-XIAO/raw/master/res/Seeeduino-XIAO-v1.0-SCH-191112.pdf

JSON file used in Arduino Studio in Additional Boards Manager URLs: https://files.seeedstudio.com/arduino/package_seeeduino_boards_index.json

Output of `pio device list':
```
C:\Users\Dawid>pio device list
COM3
----
Hardware ID: USB VID:PID=2886:802F SER=7
Description: Urządzenie szeregowe USB (COM3) <-- means "USB serial device"
```

Thanks for help

ptrm, 2020-04-22:
In the [Additional Board Manager json](https://files.seeedstudio.com/arduino/package_seeeduino_boards_index.json) it is listed along already supported [Seeeduino LoRaWAN](https://docs.platformio.org/en/latest/boards/atmelsam/seeeduino_lorawan.html) which has the same specs mcu-wise. [The pinout is obviously different](http://wiki.seeedstudio.com/Seeeduino_LoRAWAN/), but it possibly can speed up the process of supporting the XIAO.

valeros, 2020-06-03:
Resolved in https://github.com/platformio/platform-atmelsam/commit/0ed89cea0a96ce960279e34c9f4fb763e728fbcd .You can try it in the upstream version of the platform https://docs.platformio.org/en/latest/platforms/atmelsam.html#stable-and-upstream-versions

closed 2020-06-03 by valeros