platformio/platform-atmelsam #106: please add board support for the SparkFun Thing Plus

drewfish, 2020-04-22:
Hi,

Could the SparkFun SAMD51 Thing Plus be added to the list of supported boards?
https://www.sparkfun.com/products/14713

I could attempt a PR but I wouldn't really know what I'm doing. It might be tricky since the only SAMD51 boards are from Adafruit and so I might confuse things during a naive copy-paste.

gounselor, 2020-04-23:
I also would like to have that. 

Arduino Support is already available described here:

https://learn.sparkfun.com/tutorials/samd51-thing-plus-hookup-guide/setting-up-the-arduino-ide

But it would be so much cooler to be able to use this board in platform.io.

oldmath101, 2020-05-13:
yes please

drewfish, 2020-05-17:
I did some poking around, trying to get this working, and it's a bit confusing.

1. I added a `boards/sparkfun_thing_plus.json` with a bunch of changes, notably `"core": "sparkfun"` and `variant": "SparkFun_SAMD51_Thing_Plus"`
2. I think the `"core": "sparkfun"` relates to package `framework-arduino-samd-sparkfun`. The `package.json` in my installed version is has version `1.7.4`.
3. However the `url` field goes to https://github.com/sparkfun/Arduino_Boards/tree/master/sparkfun/samd, in which I'm having a hard time finding a reference to `1.7.4` so I'm not sure how that version of that package is built.

Digging around more it looks like package `framework-arduino-samd-adafruit` has a bunch of changes in it specifically for the SAMD51, changes which I don't see in my installed version of pacakge `framework-arduino-samd-sparkfun`. So, I think this project is beyond my ability at the moment, since I'm quite confused about how the different versions of the packages relate to what I see on github.

tjpeden, 2020-05-18:
@drewfish the `1.7.4` refers to the `.tar.bz2` files [here](https://github.com/sparkfun/Arduino_Boards/tree/master/IDE_Board_Manager) which have version number in their name. I was looking at this the other day to see if I could add TinyUSB support for this board. No luck with my endeavor but maybe this information will help you with yours as I would like this as well!

drewfish, 2020-05-18:
Thanks @tjpeden. That appears to be installed by the Arduino IDE since it's installed for me locally in `~/.arduino15/packages/SparkFun/hardware/samd/1.7.4/`. However I'm trying to figure out how the following was installed `~/.platformio/packages/framework-arduino-samd-sparkfun`, and what source code it was built from.

drewfish, 2020-05-18:
I was able to create a `sparkfun_thing_plus.json` and get it to _compile_ at least, however I had to do a few weird things:
* Create a `boards/sparkfun_thing_plus.json` in `~/.platformio/platforms/atmelsam/`.
* In that file specify `"variant": "SparkFun_SAMD51_Thing_Plus"`.
* In that file specify `"core": "adafruit"`.
* Copy `~/.platformio/packages/framework-arduino-samd-sparkfun/variants/SparkFun_SAMD51_Thing_Plus` into `~/.platformio/packages/framework-arduino-samd-adafruit/variants/`.

It appears that only the `framework-arduino-samd-adafruit` platformio package has support for SAMD51. When I try to use `"core": "sparkfun"` I get compile errors talking about TC2_CH1 and PIN_ATTR_PWM_G not declared. Indeed when I do a recursive diff between the `framework-arduino-samd-sparkfun` and `framework-arduino-samd-adafruit` packages I see a lot of code added to support SAMD51, as well as the following comment in a lot of headers:

> SAMD51 support added by Adafruit - Copyright (c) 2018 Dean Miller for Adafruit Industries

drewfish, 2020-05-18:
It appears that this file has both `TC2_CH1` and `PIN_ATTR_PWM_G` defined:
https://github.com/sparkfun/Arduino_Boards/blob/master/sparkfun/samd/cores/arduino51/WVariant.h

So it doesn't appear that that file is the source of version `1.7.4` of the platformio package `framework-arduino-samd-sparkfun`.

commit 00d69bca (2020-06-04): Add new board SparkFun Thing Plus // Resolve #106
closed 2020-06-04 by valeros