platformio/platform-atmelsam #108: SAMD21G18 program crashes when calling Reset_Handler

ASL07, 2020-05-15:
What kind of issue is this?
- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: OS X

**PlatformIO Version** (`platformio --version`): 4.3.3

### Description of problem

I am unable to load/start the application for SAMD21G18A (Prodino Mkr GSM). It crashes when Reset_Handler is called 
I have reflashed this bootloader https://github.com/arduino/ArduinoCore-samd/blob/master/bootloaders/mkrgsm1400/samd21_sam_ba_arduino_mkrgsm1400.hex in case it was a bootloader problem
To me, it seems there is an issue with the startup file

I tried to debug the application, and I get an error when it tries to call Reset_Handler(). 
I have also tried with a different application and the same error happens.
Please see some screenshots of the line/instruction where it crashes:

[image]

[image]

This is what I get on the JLink console:
```
Downloading 16016 bytes @ address 0x00002000
Downloading 15984 bytes @ address 0x00005E90
Downloading 15984 bytes @ address 0x00009D00
Downloading 5800 bytes @ address 0x0000DB70
Downloading 92 bytes @ address 0x0000F218
Downloading 752 bytes @ address 0x0000F274
Writing register (PC = 0x    7038)
Received monitor command: reset
Resetting target
Read 4 bytes @ address 0x00007038 (Data = 0x4A12B510)
Reading register (MSP = 0xFFFFFFFC)
Reading register (PSP = 0xB4FFF7FC)
Reading register (PRIMASK = 0x       0)
Reading register (BASEPRI = 0x       0)
Reading register (FAULTMASK = 0x       0)
Reading register (CONTROL = 0x       0)
Read 1 bytes @ address 0x00007038 (Data = 0x10)
Performing single step...
ERROR: Failed to read current instruction.
...Target halted (Vector catch, PC = 0xFFFFFFFE)
Reading all registers
WARNING: Failed to read memory @ address 0xFFFFFFFE
Read 4 bytes @ address 0x2000032E (Data = 0xB570BD01)
Reading 64 bytes @ address 0x20000300
Reading register (MSP = 0xFFFFFFFC)
Reading register (PSP = 0xB4FFF7FC)
Reading register (PRIMASK = 0x       0)
Reading register (BASEPRI = 0x       0)
Reading register (FAULTMASK = 0x       0)
Reading register (CONTROL = 0x       0)
Read 4 bytes @ address 0x2000032E (Data = 0xB570BD01)
```

valeros, 2020-05-15:
I don't this board `Prodino Mkr GSM` in the list of supported boards. Is it a clone of `mkrgsm1400`? Does it work in the Arduino IDE?

ASL07, 2020-05-15:
It is based on Arduino MKR GSM
I have fixed the issue and it is not related to Platformio.

Apparently the bootloader or fuse configuration was corrupted. I have fixed it by flashing the feather bootloader following this https://learn.adafruit.com/how-to-program-samd-bootloaders?view=all

closed 2020-05-15 by ASL07