platformio/platform-atmelsam #114: Make it possible to undefine USBCON

prices, 2020-05-29:
Right now it hard codes USBCON on the cpp command line.  I am currently using a board that doesn't have any USB connectivity.  I either have to have a custom fork of this project, or add extra modifications in to ArduinoCore-samd to have yet another check to remove USB functionality.

The patch I used to fix the problem is:

```diff
diff --git a/builder/frameworks/arduino/arduino-common.py b/builder/frameworks/arduino/arduino-common.py
index 162dcab..5368ea4 100644
--- a/builder/frameworks/arduino/arduino-common.py
+++ b/builder/frameworks/arduino/arduino-common.py
@@ -69,8 +69,7 @@ env.Append(
 
     CPPDEFINES=[
         ("ARDUINO", 10805),
-        ("F_CPU", "$BOARD_F_CPU"),
-        "USBCON"
+        ("F_CPU", "$BOARD_F_CPU")
     ],
 
     CPPPATH=[

```
That fixes my problem, but it would create a problem for many people who expect it to automatically be set.

So I am maybe thinking to put something in the board file to remove that for that particular board.  Maybe:

```json
{
    "build": {
         "nousb": true
    }
}
```

I am happy to code this up and submit a pull request, I just wanted to open this for discussion before I went and made something you guys didn't like.

valeros, 2020-05-29:
Hi @prices ! Please use [build_unflags](https://docs.platformio.org/en/latest/projectconf/section_env_build.html#build-unflags) option.

prices, 2020-05-29:
That's embarrassing, especially considering how long I have been doing it the way I posted above.

Thanks!