platformio/platform-atmelsam #121: [Feature Request] SAMC21 Arduino Support

davidcutting42, 2020-07-20:
I've done a "quick and dirty" port of the Arduino core to the SAMC21 based on the Mattairtech core. I would like to add this core to PlatformIO after cleaning it up a bit. I'm willing to put in the work needed to add it, I just need pointers on where to start.

davidcutting42, 2020-07-27:
Looking closer, it seems that @valeros has been working on implementing the mattairtech core (https://github.com/platformio/platform-atmelsam/tree/feature/mattairtech-core). @valeros, where do you need help in testing/development? I've got several SAMC boards available here.

okelk, 2020-10-28:
@davidcutting42 did this go anywhere? I would love to have support for the SAMD11 - which is also in the mattairtech-core...but I don't quite know where to start.

davidcutting42, 2020-10-28:
@okelk I made the SAMC21 work in the Arduino IDE environment, but there is still no PIO support. Seems that the PIO team doesn't really care about supporting stuff like this where there aren't many hobbyist level development boards available.

noisymime, 2021-03-01:
Wishful thinking, but any updates on this from @valeros or anyone else? 

I can understand it being a niche MCU, but I think there is still a demand out there for automotive projects and anyone looking for 5v ARM based arrangement.

valeros, 2021-03-01:
Hi guys! Sadly, but we have no plans on providing support for SAMC21 at the moment. The https://github.com/platformio/platform-atmelsam/tree/feature/mattairtech-core is an old WIP branch that is not meant to be used by users, besides it doesn't cover the SAMC21 MCUs