platformio/platform-atmelsam #123: Nano 33 IOT Compilation Error "Missing SNU.h" using ArduinoIoTCloud

sankara, 2020-08-31:
I'm getting the following error while trying to compile my sketch which has a dependency on ArduinoIoTCloud. The library SNU is part of the core files at [ArduinoCore-samd](https://github.com/arduino/ArduinoCore-samd). I was able to workaround it by manually copying the library files to my local lib folder. I'm more than happy to help fix this if you could point me in the general direction of the issue. 

```
[env:nano_33_iot]
platform = atmelsam
board = nano_33_iot
framework = arduino
```

```
In file included from .pio/libdeps/nano_33_iot/ArduinoIoTCloud/src/ArduinoIoTCloudTCP.cpp:31:0:
.pio/libdeps/nano_33_iot/ArduinoIoTCloud/src/utility/ota/OTA.h:29:12: fatal error: SNU.h: No such file or directory

*************************************************************
* Looking for SNU.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SNU.h"
* Web  > https://platformio.org/lib/search?query=header:SNU.h
*
*************************************************************

   #include <SNU.h>
            ^~~~~~~
compilation terminated.
*** [.pio/build/nano_33_iot/libff9/ArduinoIoTCloud/ArduinoIoTCloudTCP.cpp.o] Error 1
```

nciric, 2020-09-13:
I just added ArduinoIoTCloud to my project and I am getting the same error when compiling from IDE. Library manager downloaded all deps required.
I'll try doing what @sankara did to see if it would work.

nciric, 2020-09-13:
Actually, you need to update/install this target "Arduino SAMD Boards (32-bit ARM Cortex-M0+)" through "Board Manager..." menu in IDE.
Once you do SNU.h will be where expected.

chrisinabox, 2021-04-29:
Just had a very similar issue when working with the ArduinoMQQTClient and WiFiNINA libraries on the Nano33BLE. Wasn't able find Print.h for example.

In the PIO Home go to Platforms -> Updates and the make sure to update Arduino SAM libraries (and the others while you're at it).

mauedu, 2021-08-09:
I'm using Arduino MKR Wifi 1010 and I'm having the same problem after using `#include <ArduinoIoTCloud.h>. All platforms and libraries have been updated.
I was able to fix this by adding `#include <SNU.h>` to my `main.cpp`.

As mentioned by @sankara, the `SNU.h` part of the core files at ArduinoCore-samd and I suspect that the problem is the way how `SNU.h` is included in `OTA-samd.cpp`:

```c++
#if OTA_STORAGE_SNU
#  include <SNU.h>
#  include <WiFiNINA.h> /* WiFiStorage */
#endif
```

workheart, 2021-11-17:
i can confirm that compiling any code witch includes ArduinoIoTCloud inside of vsCode (using LDF) gives an error not finding SNU.h

alvarin32, 2022-08-25:
I faced the same problem with an Arduino WiFi1010 when building my code:
`
.pio\libdeps\mkrwifi1010\ArduinoIoTCloud\src\utility\ota\OTA-samd.cpp:31:12: fatal error: SNU.h: No such file or directory
`
 All platforms and libraries were updated and the error did not get solved. I followed what @mauedu wrote (i.e., _by adding #include <SNU.h> to my main.cpp._ ) and it worked.