platformio/platform-atmelsam #128: [Question] What is needed to add support to new boards/mcu/frameworks ?

chepo92, 2020-11-13:
In general, what is needed to add support to new boards/mcu/frameworks ?
So far I've read and see in PR's 
- For new boards (with already supported MCU) one may follow https://docs.platformio.org/en/latest/platforms/creating_board.html and create a new_board.json as in #73 
- For new framework a builder.py  is needed, as in #84 
- For a new atmel mcu is still unclear to me 

Any guideline? I want to work in adding support to SAM4 (and maybe others mcu) but i'm needing some docs

valeros, 2020-11-13:
Hi @chepo92 ! Adding a new MCU is pretty much the same as adding a new board as long as the CPU is ARM-based. The main question is what framework/SDK you plan to use with that MCU?

chepo92, 2020-11-13:
thanks @valeros 
So also I need  (as upported / to port) a framework for that MCU, right? I'm about to try #81, as CMSIS should support SAM4 (correct me if wrong) 

My plans are first to have a different IDE (from atmel studio/eclipse) to program SAM4 boards and then to add support in Marlin firmware to SAM4, I think i would need to write a HAL for that

Any advice would be welcome

valeros, 2020-11-13:
> So I also need (a supported / to port) a framework for that MCU, right? I'm about to try #81, as CMSIS should support SAM4 (correct me if wrong)

Well, not necessarily, you can still write some application without any HAL, but with a framework it's significantly easier. As for CMSIS build script, it requires some new re-examination to bring it back to life since a lot has changed since that PR.

chepo92, 2023-09-03:
Hi @valeros, i'm reviving this, with more time and budget :D, I pretend to enable the use with Zephyr framework I know it supports the chip (https://docs.zephyrproject.org/latest/boards/arm/sam4e_xpro/doc/index.html), and then with Arduino framework, what are the next steps I should follow? I have started reading some PR that add new boards

valeros, 2023-09-04:
Hi @chepo92, you can take a look at some board manifest that already supports those two frameworks (e.g. https://github.com/platformio/platform-atmelsam/blob/develop/boards/zero.json). AFAIK, there are no Arduino cores that support SAM4, so you're on your own here. As for Zephyr, please note that the latest supported by PlatformIO is 2.7.1, not sure if your target board was available in that release.