platformio/platform-atmelsam #131: SAMR34 , Platform IO Hex/bin/elf does not work with custom board using J-link programmer

Krunal090, 2020-12-05:
Hello Everyone ,
I am new for Platform IO.
I have SAMR34 Xplained Pro Evaluation Kit and program run perfect with inbuilt EDBG chip with platform IO with Visual code studio.
Now i will load same hex in custom board with j-link EDU from atmel studio then SAMR34 not initiated . But if I make sample code in ATMEL studio then hardware run perfect.
So let me know the solution which i have made in SAMR34 Xplained Pro Evaluation Kit and how can i load that hex file in Custom developed board to make it run.

Krunal090, 2021-01-05:
Does any one make firmware with platform IO and load this hex to making custom board ?

valeros, 2021-01-06:
Hi @Krunal090 ! What framework do you use with your `SAMR34 Xplained Pro Evaluation Kit`?

Krunal090, 2021-01-06:
@valeros , used below platform.
https://github.com/Wiz-IO/platform-sam-lora

valeros, 2021-01-06:
Please open an issue in that repository.

closed 2021-01-06 by valeros