platformio/platform-atmelsam #133: A Problem witth VS and PlatformIO IDE

RieJuerg, 2020-12-21:
Hello MaxGerhard,
As requested by you, the LOG file:

The terminal process "C:\Users\Riege\.platformio\penv\Scripts\pio.exe 'run', '--target', 'upload'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

> Executing task in folder Arduino: C:\Users\Riege\.platformio\penv\Scripts\pio.exe run --target upload <

Processing mkr1000USB (platform: atmelsam; board: mkr1000USB; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelsam/mkr1000USB.html
PLATFORM: Atmel SAM (5.0.2) > Arduino MKR1000
HARDWARE: SAMD21G18A 48MHz, 32KB RAM, 256KB Flash
DEBUG: Current (atmel-ice) External (atmel-ice, blackmagic, jlink)
PACKAGES:
 - framework-arduino-samd 1.8.9
 - framework-cmsis 1.40500.0 (4.5.0)
 - framework-cmsis-atmel 1.2.0
 - tool-bossac 1.10700.190624 (1.7.0)
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\mkr1000USB\src\main.cpp.o
Compiling .pio\build\mkr1000USB\FrameworkArduinoVariant\variant.cpp.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\Print.cpp.o
Archiving .pio\build\mkr1000USB\libFrameworkArduinoVariant.a
Compiling .pio\build\mkr1000USB\FrameworkArduino\Reset.cpp.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\SERCOM.cpp.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\USB\CDC.cpp.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\USB\PluggableUSB.cpp.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\USB\USBCore.cpp.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\USB\samd21_host.c.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\Uart.cpp.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\WInterrupts.c.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\WString.cpp.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\abi.cpp.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\avr\dtostrf.c.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\cortex_handlers.c.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\delay.c.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\hooks.c.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\itoa.c.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\main.cpp.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\new.cpp.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\pulse.c.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\pulse_asm.S.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\startup.c.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\wiring.c.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\wiring_analog.c.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\wiring_digital.c.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\wiring_private.c.o
Compiling .pio\build\mkr1000USB\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\mkr1000USB\libFrameworkArduino.a
Linking .pio\build\mkr1000USB\firmware.elf
Building .pio\build\mkr1000USB\firmware.bin
Checking size .pio\build\mkr1000USB\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   7.2% (used 2364 bytes from 32768 bytes)
Flash: [          ]   4.3% (used 11212 bytes from 262144 bytes)
Configuring upload protocol...
AVAILABLE: atmel-ice, blackmagic, jlink, sam-ba
CURRENT: upload_protocol = sam-ba
Looking for upload port...
Auto-detected: COM3
Uploading .pio\build\mkr1000USB\firmware.bin
No device found on COM3
*** [upload] Error 1
======================================================================================== [FAILED] Took 12.59 seconds ========================================================================================
The terminal process "C:\Users\Riege\.platformio\penv\Scripts\pio.exe 'run', '--target', 'upload'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

PlatformIO.INI:

[env:mkr1000USB]
platform = atmelsam
board = mkr1000USB
framework = arduino
board_upload.use_1200bps_touch = no 
board_upload.wait_for_upload_port = no

valeros, 2021-01-06:
Hi @RieJuerg ! What is the reason for using custom upload settings?

RieJuerg, 2021-01-07:
hello valeros,
thank you very mutch for your replay.
I have a problem executing an Arduino sketch via Visual Studio on an Arduino board MKR1000. If I upload the sketch and specify Port6, it will be changed to Port3. When trying to upload the sketch to Port3, an error message appears.
I have set the upload Port6 in platformIO.ini at Port6, because the correct upload Port is Port6. I cannot understand why Visual Studio is changing the port to port 3.
regards Jurgen

closed 2021-01-07 by RieJuerg