platformio/platform-atmelsam #136: **Development Platform or Board** request: RK002 Retrokits Board

m-r-m-s, 2020-12-23:
What kind of issue is this?

- [x] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform
*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

Hi, this is a request to add the Retrokits RK-002 board to platformio. It is already supported by the Arduino IDE:

https://www.retrokits.com/rk002/arduino/package_retrokits_index.json

More info here: 
https://www.retrokits.com/rk002-duy/

Thank you

bortch, 2023-11-08:
Hello,
I'm also interested in adding the RK002 to platformio. Is this possible using the method described for [adding custom boards](https://docs.platformio.org/en/stable/platforms/creating_board.html)?

valeros, 2023-11-08:
@bortch Unfortunately, these boards require additional packages which are not available in the PlatformIO registry.