platformio/platform-atmelsam #139: SAMD Sodaq Sara board files giving LITTLE_ENDIAN re-definition warnings.

gmos, 2021-01-04:
See below. Is this for you or should I discuss with Sodaq??

```
In file included from C:\Users\GijsMos\.platformio\packages\framework-cmsis-atmel\CMSIS\Device\ATMEL/samd21/include/samd21.h:83:0,
                 from C:\Users\GijsMos\.platformio\packages\framework-cmsis-atmel\CMSIS\Device\ATMEL/samd.h:117,
                 from C:\Users\GijsMos\.platformio\packages\framework-cmsis-atmel\CMSIS\Device\ATMEL/sam.h:470,
                 from C:\Users\GijsMos\.platformio\packages\framework-arduino-samd-sodaq\cores\arduino/WVariant.h:22,
                 from C:\Users\GijsMos\.platformio\packages\framework-arduino-samd-sodaq\variants\sodaq_sara/variant.h:44,
                 from C:\Users\GijsMos\.platformio\packages\framework-arduino-samd-sodaq\libraries\Wire/Wire.h:24,
                 from src\Maker.cpp:16:
C:\Users\GijsMos\.platformio\packages\framework-cmsis-atmel\CMSIS\Device\ATMEL/samd21/include/samd21j18a.h:230:0: warning: "LITTLE_ENDIAN" redefined
 #define LITTLE_ENDIAN          1

In file included from c:\users\gijsmos\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\sys\types.h:67:0,
                 from c:\users\gijsmos\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\stdio.h:61,
                 from C:\Users\GijsMos\.platformio\packages\framework-arduino-samd-sodaq\cores\arduino/Print.h:23,
                 from C:\Users\GijsMos\.platformio\packages\framework-arduino-samd-sodaq\cores\arduino/Stream.h:26,
                 from C:\Users\GijsMos\.platformio\packages\framework-arduino-samd-sodaq\libraries\Wire/Wire.h:23,
                 from src\Maker.cpp:16:
c:\users\gijsmos\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\machine\endian.h:17:0: note: this is the location of the previous definition
 #define LITTLE_ENDIAN _LITTLE_ENDIAN
```

valeros, 2021-01-04:
Hi @gmos ! We prefer to show warnings instead of suppressing them. If this doesn't work for you, please add `-w` flag to the `build_flags` option in your `platformio.ini` as it done in the Arduino SAMD by default.

closed 2021-01-04 by valeros

gmos, 2021-01-04:
Question was not to suppress the warnings, but to get the board files fixed.
Whom is responsible for these files. Perhaps I can re-submit there. Where do you pull them from?

valeros, 2021-01-04:
Here is the Sodaq SAMD repository https://github.com/SodaqMoja/SodaqCore-samd