platformio/platform-atmelsam #148: Support for ATSAMD11 boards

Theremingenieur, 2021-02-25:
What kind of issue is this?

- [x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

I'd like to see support for ATSAMD11 boards added to platformIO. Although not very popular, these little beasts come in a hand solder friendly SOIC-14 format and have native USB. There is already a fork of the Arduino-samd core by mattairtech which adds everything [https://github.com/mattairtech/ArduinoCore-samd] but I have no clue how to "import" these additions into platformIO.

Thank you in advance for your kind consideration !

Theremingenieur, 2021-02-27:
I discover only now that a similar request has already been made by @ivankravets in #28.
The Mattairtech fork of the Arduino samd cores adds not only support for the Mattairtech dev boards, but also for the generic SAMD11C14.

Theremingenieur, 2021-04-14:
Since the MattairTech cores seem to be orphaned since 2018, it might be better to integrate @wagiminator's [more recent SAMD cores](https://github.com/wagiminator/SAMD-Development-Boards):

harry-finch, 2022-12-26:
@Theremingenieur May I ask how you've been using the Mattairtech or even @wagiminator's core in platformio? I have installed the core but the D11C or D11D "boards" are not available in the menu when creating a new project...

There's the core from Quenting Bolsee, which is a fork from Mattairtech, which works well in the Arduino IDE but which I haven't been able to use in Platformio. https://github.com/qbolsee/ArduinoCore-fab-sam