platformio/platform-atmelsam #149: Version 6.0.0 does not compile for Seeed XIAO with SPI

savejeff, 2021-03-02:
Yesterday I updated to the 6.0.0 version on PlatformIO
Using SPI gives me a compile error: 
[image]

using this simple code:

`#include <Arduino.h>

#include "SPI.h"

void setup() {
  SPI.begin();
  
}

void loop() {
 
}`

valeros, 2021-03-02:
Thanks for reporting. Should be fixed in the development branch, please try to compile your project using the following configurtion:
```
[env:seeed_xiao]
platform = https://github.com/platformio/platform-atmelsam.git
board = seeed_xiao
framework = arduino
```

closed 2021-03-02 by valeros

savejeff, 2021-03-02:
Oh i didn't knwo you could just use the github link to the platform code. Nice to know. 

For now, I will just downgrade to 5.1.1 in PlatformIO and wait for a new version

commit c0a37c6f (2021-03-02): Fix linker flag for Seeed Core // Resolve #149