platformio/platform-atmelsam #150: [Seeed Wio Terminal] undefined reference to `__wrap__write'

matsujirushi, 2021-03-03:
I got below error using platform-atmelsam v6.0.0.

```
Linking .pio\build\seeed_wio_terminal\firmware.elf
c:/users/takashi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m/fpv4-sp/hard\libc_nano.a(lib_a-writer.o): In function `_write_r': 
writer.c:(.text._write_r+0x10): undefined reference to `__wrap__write'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\seeed_wio_terminal\firmware.elf] Error 1
```

https://github.com/SeeedJP/wioterminal-aziot-example/issues/5

valeros, 2021-03-03:
Duplicate https://github.com/platformio/platform-atmelsam/issues/149. Fixed in the development branch:
```
[env:seeed_xiao]
platform = https://github.com/platformio/platform-atmelsam.git
board = seeed_xiao
framework = arduino
```

closed 2021-03-03 by valeros

kartben, 2021-03-04:
Ah, the usual suspects! Just had this issue -- cool to see it's already fixed :)