platformio/platform-atmelsam #151: Uploading to Arduino MKR Zero (ABX00012) from KVM VM does not work

pboguslawski, 2021-03-09:
Hello,

Trying to upload any program from PlatformIO with Atmel SAM 6.0.1 running inside KVM virtual machine (Debian 10 64-bit) to Arduino MKR Zero (redirected USB device from host to virtual machine) throws an error:

```
[...]
Auto-detected: /dev/ttyACM0
Forcing reset using 1200bps open/close on port /dev/ttyACM0
Waiting for the new upload port...
Uploading .pio/build/mkrzero/firmware.bin
No device found on ttyACM0
*** [upload] Error 1
```

USB device is visible in VM:

```
$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 2341:804f Arduino SA 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

No such problem with Arduino Pro Mini nor Nano Every in the same VM.

No such problem with Arduino MKR Zero when uploading from PlatformIO with Atmel SAM 6.0.1 running on same physical host (not in VM).

Regards,
Paweł

pboguslawski, 2021-03-10:
Board does not change its IDs; on host when mkrzero is executing blink:

```
$ lsusb | grep -i arduino
Bus 002 Device 014: ID 2341:804f Arduino SA 
```
on host when mkrzero is in bootloader mode (after 2x fast reset presses, when led is indicating bootloader mode correctly /dimming/):

```
$ lsusb | grep -i arduino
Bus 002 Device 015: ID 2341:004f Arduino SA 
```

When the problem occurs in VM, USB redirection is stable (in case of USB ID changes redirection would be broken). So the problem source seems to be different than USB ID changes.

valeros, 2021-03-10:
> Board does not change its IDs; on host when mkrzero is executing blink:

Hm, I see here two different PIDs `804f` and `004f`. You use both devices are redirected?
Anyway, could you please try to upload a simple blink using Arduino IDE in VM?

pboguslawski, 2021-03-10:
> could you please try to upload a simple blink using Arduino IDE in VM

Will try but need more time (Debian 10 in VM has older Arduino-IDE packaged that does not support mkrzero).

pboguslawski, 2021-03-10:
Just checked that in Arch linux with kernel 5.11.4-arch1-1 (host & vm) on different computer that uploading using Arduino IDE 1.8.13 does not work either. Same error.

Seems it's not PlatformIO fault but some kind of incompatibility between mkrzero USB and USB redirection in VM.

Thank you for help!

valeros, 2021-03-10:
As I mentioned above, mkrzero in regular and bootloader mode represents two different devices, you need to configure your VM's USB filters so the board in both modes is automatically attached to your VM.

pboguslawski, 2021-03-10:
Changing IDs is not a cause of this problem - see

https://github.com/platformio/platform-atmelsam/issues/151#issuecomment-795339574