platformio/platform-atmelsam #154: SERCOM Peripherals not updating

MasterTim17, 2021-03-18:
Hello im currently debugging a SAMD21G18A and found out, that the SERCOM Peripherals tab doesnt change values. Other peripherals like PORT or PM->APBCMASK change value. The values of sercom always stay 0, while watching them via watch `((Sercom*)0x42000c00).SPI.DATA.bit.DATA` reveals the correct value

valeros, 2021-03-19:
Hi @MasterTim17 ! That's strange, what's the version of the `atmelsam` platform installed in your system?

MasterTim17, 2021-03-23:
im using version 6.0.1

valeros, 2021-03-23:
Could you please try the upstream version of the platform, for example:
```ini
[env:test]
platform = https://github.com/platformio/platform-atmelsam.git
framework = ...
board = ...
```

MasterTim17, 2021-03-23:
Tried it, same problem